genodelabs/genode #3671: regression: Sculpt only shows last AHCI device

NobodyIII, 2020-02-29:
With the new 20.02 release, only the last device connected to the AHCI controller is visible in Sculpt. The other SATA devices aren't shown in either the GUI or in the block_devices report, despite being listed in the log. This is a regression, as the devices were detected and listed properly in version 19.11.

chelmuth, 2020-03-02:
I can confirm the behavior with the commit above.
```
> make -C build/x86_64 run/ahci_bench K=nova B=pc
[init -> ahci_drv] number of ports: 6 pi: 0x3f
[init -> ahci_drv] Warning: CMD.ST bit set during device reset --> unknown behavior
[init -> ahci_drv] 		#0: ATA
[init -> ahci_drv] 		#1: off (ATAPI)
[init -> ahci_drv] Warning: CMD.ST bit set during device reset --> unknown behavior
[init -> ahci_drv] 		#2: ATA
[init -> ahci_drv] 		#3:  off (unknown device signature)
[init -> ahci_drv] 		#4:  off (unknown device signature)
[init -> ahci_drv] 		#5:  off (unknown device signature)
[init -> report_rom] report 'ahci_drv -> ports'
[init -> report_rom]   <ports>
[init -> report_rom]   	<port num="0" type="ATA" block_count="32768" block_size="512" model="QEMU HARDDISK" serial="QM00013"/>
[init -> report_rom]   </ports>
[init -> report_rom] report 'ahci_drv -> ports'
[init -> report_rom]   <ports>
[init -> report_rom]   	<port num="2" type="ATA" block_count="32768" block_size="512" model="QEMU HARDDISK" serial="QM00017"/>
[init -> report_rom]   </ports>
[init -> test-ahci] Warning: block count 32768 size 512
```

It seems two reports are generated consecutively but each reports contains only 1 device.

chelmuth, 2020-03-03:
Merged the fix to staging. @NobodyIII can you confirm that it fixes your issue?

NobodyIII, 2020-03-04:
That commit fixed that problem, but now part_blk is stalling on the devices that used to be hidden.

chelmuth, 2020-03-04:
> That commit fixed that problem, but now part_blk is stalling on the devices that used to be hidden.

Is this a regression or did the device also not work with 19.11?

ssumpf, 2020-03-04:
> > That commit fixed that problem, but now part_blk is stalling on the devices that used to be hidden.
> 
> Is this a regression or did the device also not work with 19.11?

In case it is a regression it would be helpful to have a synthetic test scenario that reproduces the behaviour.

nfeske, 2020-03-04:
@NobodyIII It is hard to diagnose the problem based on only the information that it gets stuck. This may be (1) an AMD-specific problem of the AHCI driver, or (2) a problem with the AHCI driver handing more than one session at a time, or (3) a regression in part_block. I think that (3) is unlikely because part_block does not even know / care about the provider of its block session.

I'm afraid that investigating (1) is not possible for us because of the lack of test hardware.

However, (2) should be possible to investigate using Qemu by connecting two virtual disks. @ssumpf Could you please give it try to attach two (different) partitioned disk images to Qemu, hand out each to a dedicated instance of part_block, and see what each instance has to report? You can do the latter by routing the generated reports to a report_rom server configured with `verbose="yes"`.

Should the issue be indeed (2) we should better fix it before publishing the official Sculpt image.

@NobodyIII to better understand the "getting stuck" situation, it would help to know the actual components running and their sessions. To get this information, can you please take the following steps (in the inspect window of Sculpt)?

1. Add a `requested="yes"` attribute to the <report> node of /config/managed/runtime. This will prompt the runtime init instance to include session information in its state report.
2. Look at /report/runtime/state via vim. The report should now contain information about all sessions requested by each component.
3. Post a copy of the runtime state report. So we can see which part_block instances are running and how their session-creation status looks like.

ssumpf, 2020-03-04:
@nfeske: Regarding the test scenario, will do.

ssumpf, 2020-03-05:
@nfeske: I have tested with two ATA images, two partitions each, one GPT one MBR disk, two block testers and two `test-block-client`s, each routed to different partitions. The tests run successfully. The generated reports also look good:

```
report 'ahci_drv -> ports'                                                                                                                                                                    
<ports>                 
  <port num="0" type="ATA" block_count="2097152" block_size="512" model="QEMU HARDDISK" serial="QM00013"/>
  <port num="1" type="ATA" block_count="4194304" block_size="512" model="QEMU HARDDISK" serial="QM00015"/>
</ports>           

report 'part_block -> partitions'
<partitions type="mbr">   
  <partition number="1" start="2048" length="1048576" block_size="512" type="131"/>
  <partition number="2" start="1050624" length="1046528" block_size="512" type="131"/>
</partitions> 
report 'part_block2 -> partitions'
<partitions type="gpt" total_blocks="4194304" gpt_total="4194237" gpt_used="4192223">
  <partition number="1" name="Linux filesystem" type="0fc63daf-8483-4772-8e79-3d69d8477de4" guid="cfd3724b-d84c-4699-8a96-630f719096f5" start="2048" length="2097152" block_size="512"/>
  <partition number="2" name="Linux filesystem" type="0fc63daf-8483-4772-8e79-3d69d8477de4" guid="1c42f041-187e-4ff2-882c-f9851712b109" start="2099200" length="2095071" block_size="512">
</partitions>
```

nfeske, 2020-03-05:
Thanks for testing. I think that this rules our the potential issue (2). Consequently, I would not no longer regard it as a showstopper for publishing the new Sculpt image.

@ssumpf do you think that it is possibile to add this particular test to our automated tests without much trouble? If yes, I would very much appreciate you adding it (preferably integrated in one of the existing run scripts).

ssumpf, 2020-03-05:
> Thanks for testing. I think that this rules our the potential issue (2). Consequently, I would not no longer regard it as a showstopper for publishing the new Sculpt image.
>
> @ssumpf do you think [...]

Sure, maybe `block_tester.run`?

NobodyIII, 2020-03-06:
I examined the issue, and it seems to be tied to the AHCI driver rather than part_blk, because I can successfully use one of the problematic SSDs in QEMU with a slightly modified version of the sculpt_test script.

ssumpf, 2020-03-06:
@nfeske: With 4f6a325 I have expanded the `block_tester.run` script to test partitions on multiple disks with multiple clients. It supports Qemu, Linux, and if one is brave enough x86 AHCI hardware. I disabled hardware for the autopilot, this way the script could be used for nightly testing on Qemu and could alert at least for some regressions in `ahci_drv` and `part_block`. What do you think?

nfeske, 2020-03-06:
I think this looks very good!

@cnuke what is your opinion about changing block_tester.run in this way?

cnuke, 2020-03-06:
@nfeske LGTM - I guess the test should be added to the `autopilot.list`.

ssumpf, 2020-03-09:
I just noticed, because I changed the output message of `test-block-client` the recipes for the block tests need to be adjusted. Fixup will follow.

ssumpf, 2020-03-09:
@chelmuth: Please also consider 4f6a325 and 54a34c5.

chelmuth, 2020-03-11:
Merged both commits to staging.

commit 4f59a319 (2020-03-03): ahci: report ports correctly

Only use one report for devices report, not one per device

fixes #3671
commit c856ba2a (2020-03-09): ahci: report ports correctly

Only use one report for devices report, not one per device

fixes #3671
closed 2020-03-26 by nfeske