genodelabs/genode #3672: Add shared component for making arm PSCI calls.

tworaz, 2020-02-29:
It'd be nice to have a shared component that armv8 platforms can use in order to make PSCI calls.

skalk, 2020-03-02:
Related to issue #3673

skalk, 2020-03-24:
@tworaz now, I had the time to review your commit. Thank you for your nice work. Beside some very few  coding style  corners, I've changed the way how psci.h included psci_call.h, and moved the 32/64-bit call header into the src/include/hw path. The reason is that all headers in src/include/hw are independent from code in src/core or src/bootstrap, and this shouldn't be changed.
Please have a look at the resulting commit e9f58d9, and give me a sign, whether you like it or not.

tworaz, 2020-03-24:
@skalk Thanks for taking the time to review and improve the code. The new version looks much better.

nfeske, 2020-03-26:
Merged to staging.

commit b51defab (2020-03-26): hw: Add interface for making basic ARM PSCI calls.

So far this only exposes two functions allowing the code to turn on
additional CPUs. There is much more functionality defined by PSCI spec
but so far its not needed by Genode.

Fixes #3672
commit dad1de18 (2020-04-17): hw: Add interface for making basic ARM PSCI calls.

So far this only exposes two functions allowing the code to turn on
additional CPUs. There is much more functionality defined by PSCI spec
but so far its not needed by Genode.

Fixes #3672
closed 2020-04-17 by chelmuth