genodelabs/genode #3673: Add support for Qemu arm virt machines to base-hw

tworaz, 2020-02-29:
My long term goal is to add VirtIO driver framework and nic, input and framebuffer VirtIO drivers to Genode. The x86_64 qemu machine can use VirtIO devices out of the box. The situation on arm is a bit more complex and VirtIO devices can only be exposed for custom "virt" machines. The goal here is to allow Genode using base-hw kernel to boot on such "virt" machines.

tworaz, 2020-02-29:
For the record, this patch needs PSCI component from #3672 to work.

skalk, 2020-03-02:
@tworaz thanks for providing both commits, it is pretty much welcome. I'll soon review your contribution.

tworaz, 2020-03-02:
@skalk Please take your time, I'm not in a hurry. The VirtIO driver support for Genode which is my stated goal is more or less in a usable state already. The code needs some cleanup before I'll push it for review, however. The slightly outdate branch containing all my work so far can be seen here: https://github.com/tworaz/genode/commits/virtio.

skalk, 2020-03-24:
@tworaz Ok, I reviewed this commit too. Beside the adaption needed to meet the minor change of the psci interface, I renamed arm_virt, as well as arm_64_virt to virt_qemu. The reason is that we already used that scheme for e.g. the zynq_qemu platform, and there are some virt_qemu solutions already flying around in some topic branches of some team members.
Thereby, I have put together the 32 and 64-bit versions as far as it is possible, and resigned to use different board names. Because the architecture reflects already the differences, but all peripherals can be the same in the future. The good thing about it is that we can spare some code duplications e.g. for the base-hw-virt_qemu package etc.
I hope you like the adaptions, which can be seen in commit 0fe6918. If you are ok with it, I would strive toward bringing it to staging. Thank you for your contribution!

tworaz, 2020-03-24:
@skalk Thank you very much for taking your time to look at this.

The fun part is I did actually call this new platform qemu_virt at some point. AFAIR I switched to arm_virt and arm_64_virt after I've decided to support both ARMv8 and ARMv7 and while at it to also add RISC-V version. I think RISC-V board name ended up being rv64_virt. I still have that code stashed on a branch somewhere, but I'll probably won't have the time to get it into good enough shape for contributing to Genode. I basically lost interest in it after getting VirtIO nic driver working. Long story short qemu_virt does look fine to me. I'll need to make some adjustments to my work branch, but this is expected.

Please feel free to land your version on staging branch whenever you wish :)

nfeske, 2020-03-26:
Merged to staging.

commit 2d04af4c (2020-03-26): hw: Add support for Qemu virt arm platforms.

This patch adds support for booting base-hw kernel on qemu-arm virt
machines. The arm_virt machine has 2GB of RAM, 2 Cortex A15 cores and
uses GICv2 interrupt controller. The arm_64_virt machine also has 2GB of
RAM, but has 4 Cortex A53 cores and uses GICv3. Both machines use PSCI
to boot additional CPU cores.

Fixes #3673
commit 585c4b8c (2020-04-17): hw: Add support for Qemu virt arm platforms.

This patch adds support for booting base-hw kernel on qemu-arm virt
machines. The arm_virt machine has 2GB of RAM, 2 Cortex A15 cores and
uses GICv2 interrupt controller. The arm_64_virt machine also has 2GB of
RAM, but has 4 Cortex A53 cores and uses GICv3. Both machines use PSCI
to boot additional CPU cores.

Fixes #3673
closed 2020-04-17 by chelmuth