genodelabs/genode #3676: timespec_get symbol is missing in libc

tworaz, 2020-02-29:
FreeBSD libc which genode uses does support it and the code is even built on Genode and exposed via libc headers. Unfortunately its not listed in the symbols file for libc so code trying to use timespec_get will fail to link.

nfeske, 2020-03-06:
@tworaz thank you for the patch. Good to see you on the issue tracker. :-)

I will cherry-pick your commit to our staging branch by the mid of next week. Currently, we have frozen the staging branch regarding any changes that do not affect the immediate Sculpt OS release (planned for Tuesday).

nfeske, 2020-03-10:
I merged your patch to staging now.

tworaz, 2020-03-10:
Thanks! It'll make my life a bit easier :)

closed 2020-03-10 by tworaz

nfeske, 2020-03-10:
BTW, it's fine to keep issues open. Since I annotated the commit message with the issue number, the issue will be closed automatically once the commit enters the master branch.

commit dfae4a29 (2020-03-10): libc: Expose timespec_get symbol.

The function is exposed in libc headers provided by Genode, the code for
the function is being compiled and actually works, but the symbol is
missing from the symbols file resulting in linking failures. Add it to
the libc symbols file.

Fixes #3676
Fixes #3677
commit 139f3903 (2020-03-26): libc: Expose timespec_get symbol.

The function is exposed in libc headers provided by Genode, the code for
the function is being compiled and actually works, but the symbol is
missing from the symbols file resulting in linking failures. Add it to
the libc symbols file.

Fixes #3676
Fixes #3677