genodelabs/genode #3681: nic_router: fix lifetime of DHCP clients

m-stein, 2020-03-04:
The NIC router currently runs into an uncaught exception when a DHCP client timeout of an interface from a period with dynamic IP config triggers after the interface was re-configured to have a static IP config. The same happens when it receives DHCP reply packets at an interface that has a static IP config (which can also occur due to reconfiguration). The general problem is that the lifetime of a DHCP client is the same as that of the interface that holds it. The lifetime of the DHCP client should be bound to the type of IP config of the interface instead.

m-stein, 2020-03-06:
@chelmuth Thanks for your review!
I've updated the originbal commit and added another one, both tested with my sculpt + VM + switching uplink:
1c724806dd nic_router: don't call DHCP client on requests
11fef0c351 nic_router: fix lifetime of DHCP clients

m-stein, 2020-03-12:
@chelmuth I added another commit that adds a simple test for the dhcp renew handshake:
766db9daab ping_nic_router.run: test dhcp renew

chelmuth, 2020-03-12:
Merged as 672cb57.

commit 4244aba6 (2020-03-11): nic_router: don't call DHCP client on requests

Until now, the DHCP client was called also for DHCP requests when an interface
had a domain but yet no IP config. Now, an interface distinguishes between DHCP
requests and replies first and then accordingly calls the DHCP server or the
DHCP client if they're available. This also prevents that the DHCP client has
to handle packet headers other than that of DHCP.

Fixes #3681
closed 2020-03-26 by nfeske