genodelabs/genode #3682: Page fault in ssh_terminal

nfeske, 2020-03-06:
On my quest of removing Noux from our run scripts, I came across ssh_terminal.run, which is currently broken:

```
build/x86_64$ RUN_OPT=--autopilot make run/ssh_terminal KERNEL=nova BOARD=pc
...
[init -> ssh_terminal] INFO: NET: Registered protocol family 16
[init -> ssh_terminal] INFO: NET: Registered protocol family 2
[init -> ssh_terminal] queue_delayed_work: queue_delayed_work called (from 0x13c1c77) return 0
[init -> ssh_terminal] queue_delayed_work: queue_delayed_work called (from 0x13dc3ea) return 0
[init -> ssh_terminal] INFO: Hash tables configured (established 524288 bind 524288)
[init -> ssh_terminal] Received mac: 52:54:00:12:34:56
[init -> ssh_terminal] Using DHCP for interface configuration.
[init -> ssh_terminal] queue_delayed_work: queue_delayed_work called (from 0x13da733) return 0
[init -> ssh_terminal] NOTICE: Sending DHCP requests ., OK
[init -> ssh_terminal] my address is 10.0.2.15
[init -> ssh_terminal] queue_delayed_work: queue_delayed_work called (from 0x13da733) return 0
[init -> ssh_terminal] queue_delayed_work: queue_delayed_work called (from 0x13da733) return 0
[init -> ssh_terminal] INFO: IP-Config: Complete:
[init -> ssh_terminal] INFO:      device=eth0, len=6 hwaddr=52:54:00:12:34:56, ipaddr=10.0.2.15, mask=255.255.255.0 , gw=10.0.2.2
spawn sshpass -p xuon ssh -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -l noux localhost -p 5555
[init -> ssh_terminal] INFO:      host=1451224I4, domain=, nis-domain=
[init -> ssh_terminal] INFO:      bootserver=10.0.2.2, rootserver=10.0.2.2, rootpath=
[init -> ssh_terminal] INFO:      nameserver0=10.0.2.3
[init -> ssh_terminal] using the pipe libc plugin
[init -> ssh_terminal] Allowed auth methods: password 
[init -> ssh_terminal] Listen on port: 22
[init -> ssh_terminal] Login configured: user charlie: password 
[init -> ssh_terminal] Login configured: user noux: password 
[init -> ssh_terminal] --- SSH terminal started ---
page fault, pd='init -> ssh_terminal' thread='pthread.1' cpu=0 ip=0xaf64431 address=0xa0701678 stack pointer=0xa04feb90 qualifiers=0x4 irUwp reason=1
^CMakefile:323: recipe for target 'run/ssh_terminal' failed
make: *** [run/ssh_terminal] Error 130
```

The fault address corresponds to the libc at offset 0x151431 in the following code:

```
/plain/no/genode.git/repos/libports/src/lib/libc/poll.cc:102 (discriminator 1)
  15141e:       48 8b bc cc 90 00 00    mov    0x90(%rsp,%rcx,8),%rdi
  151425:       00 
  151426:       48 0f a3 d7             bt     %rdx,%rdi
  15142a:       73 05                   jae    151431 <__sys_poll+0x211>
/plain/no/genode.git/repos/libports/src/lib/libc/poll.cc:103
                        fds[i].revents |= POLLOUT;
  15142c:       66 83 4b 06 04          orw    $0x4,0x6(%rbx)
/plain/no/genode.git/repos/libports/src/lib/libc/poll.cc:105
                }
                if (FD_ISSET(fd, &exceptfds)) {
  151431:       48 8b 8c cc 10 01 00    mov    0x110(%rsp,%rcx,8),%rcx
  151438:       00 
  151439:       48 0f a3 d1             bt     %rdx,%rcx
  15143d:       73 05                   jae    151444 <__sys_poll+0x224>
/plain/no/genode.git/repos/libports/src/lib/libc/poll.cc:106
                        fds[i].revents |= POLLERR;
  15143f:       66 83 4b 06 08          orw    $0x8,0x6(%rbx)
/plain/no/genode.git/repos/libports/src/lib/libc/poll.cc:94 (discriminator 2)
```

The fault occurs in the stack area but on a different stack than the faulter. I suspect that the de-referenced object lies on the application stack whereas the fault occurs inside the libc kernel where the kernel stack is active. It looks like a huge object is placed somewhere on the application stack. The application-stack usage is 0xa07fffff - 0xa0701678 = 1042823.

chelmuth, 2020-03-06:
By carefully rewinding the patches to *repos/gems/src/server/ssh_terminal* I discovered eaefcc2c6fc7eae4ae2ab42a0a00099d6fe00bf6 introduces this regression. From my experience, I doubt the bug can be triggered in older version as I used a previous version for some time in combination with our Genode-based buildbot. This does not exclude that it already slumbered in the background and was only activated by other fixes in the commit (like correct connection notification)!

nfeske, 2020-04-24:
Because of this problem, the ssh_exec_channel.run script still fails on a number of platforms including okl4/x86_32, pistachio/x86_32, sel4/x86_32, nova/x86_32, nova/x86_64.

cnuke, 2020-11-17:
Commits a2d7b3b, 53d6dd8 and b33e007 *should* address the current test failures with `ssh_exec_channel.run`.

@chelmuth please merge to staging.

nfeske, 2020-11-18:
Thanks a lot @cnuke!

cnuke, 2020-11-19:
Apparently the page-fault is fixed but now the test gets stuck on all kernels but NOVA, sometimes exhibiting
```
[init -> ssh_terminal] Error: deadlock ahead, mutex=0x3d005c, return ip=0x10dd30ed
```
on OKL4 and seL4.

As the component is not actively used by us, not the particular _exec channel_ feature anyway, and at least I currently lack the time to look more deeply into the issue I would rather consider disabling the test. Any thoughts?

chelmuth, 2020-11-20:
In my opinion, we should not keep the component in the genode repo if it's not tested, but move it to world.

alex-ab, 2020-11-27:
> Apparently the page-fault is fixed but now the test gets stuck on all kernels but NOVA, sometimes exhibiting
> 
> ```
> [init -> ssh_terminal] Error: deadlock ahead, mutex=0x3d005c, return ip=0x10dd30ed
> ```
> 
> on OKL4 and seL4.

With commit ac3c6ae7728e2e45eabb27d64630a5c9b74e57d9 the deadlock does not happen anymore.

nfeske, 2020-11-27:
Thanks @alex-ab, what a pleasant surprise! I merged your fix to staging.

nfeske, 2021-02-26:
The page fault is gone.

closed 2021-02-26 by nfeske