genodelabs/genode #3683: vbox5/seoul: adding test VM causing significant stress nightly

alex-ab, 2020-03-06:
In order to discover performance variations and functional regressions more early, some VMs causing some defined load in our testbed would be nice. Current tests are mainly testing successful boot and shutdown and shared folder interaction. Test VMs with a variable number of vCPUs and load to utilize them are missing currently. Measurements about the runtime and the actual cpu load of the runs are desired to detect variations earlier.

alex-ab, 2020-03-06:
A first state is available. A VM (32/64bit) boots up, loads via tftp a script and does what it got told. The current the script clones the Genode sources, prepares build and some contrib code (nova & libc) and build a libc application. The script makes via ttyS0 some diagnostic output, which are used to measure the elapsed time on host. Actually, both, the guest and host, do the measurement, so we can detect also time measure anomalies in case they happen.

Several runs are done with variable vCPU count. Finally the results are printed and compared against the last values of the test machine. The output currently looks like this:

```
...
[init -> log_terminal] cpus=1 - clone=13 - prepare_port=45 - make=81
[init -> log_terminal] cpus=2 - clone=10 - prepare_port=40 - make=49
[init -> log_terminal] cpus=1 - clone=13 - prepare_port=38 - make=78
[init -> log_terminal] cpus=2 - clone=11 - prepare_port=36 - make=49
[init -> log_terminal] cpus=3 - clone=10 - prepare_port=38 - make=39
[init -> log_terminal] cpus=4 - clone=10 - prepare_port=41 - make=34

(host)                 cpus=1 - clone=13 - prepare_port=45 - make=81 boottimeVM=20 - VMM generic
(host)                 cpus=2 - clone=10 - prepare_port=40 - make=49 boottimeVM=21 - VMM generic
(host)                 cpus=1 - clone=13 - prepare_port=38 - make=78 boottimeVM=25 - VMM NOVA specific
(host)                 cpus=2 - clone=11 - prepare_port=36 - make=49 boottimeVM=25 - VMM NOVA specific
(host)                 cpus=3 - clone=10 - prepare_port=38 - make=39 boottimeVM=26 - VMM NOVA specific
(host)                 cpus=4 - clone=10 - prepare_port=41 - make=35 boottimeVM=26 - VMM NOVA specific
 
host     results 1 13 45 81 2 10 40 49 1 13 38 78 2 11 36 49 3 10 38 39 4 10 41 35
guest    results 1 13 45 81 2 10 40 49 1 13 38 78 2 11 36 49 3 10 38 39 4 10 41 34
expected results 1 13 41 80 2 10 42 50 1 13 36 78 2 11 35 48 3 9 39 39 4 10 41 33
boottime host    20 21 25 25 26 26
boottime expect  21 21 26 25 26 27
1 results deviate too much

Run script execution successful.
```

As described in the beginning, the cpu load and Seoul part is currently missing and is planned to work on.

chelmuth, 2020-03-11:
@alex-ab that's even better as I expected from the concept you described. @cnuke we should not wait too long to record the data in the CI database and add a new graph to the front page.

alex-ab, 2020-04-08:
> @alex-ab that's even better as I expected from the concept you described. @cnuke we should not wait too long to record the data in the CI database and add a new graph to the front page.

I adjusted the CI database and added graphs to the front page. @chelmuth, @nfeske: may you please add the commits 4e991a82b4556825c3b679d826d97c37c36f34b1 78b2c69af285c75d0f46f913e4d637c5fd71b6fc a98777ffbcb570c5ffd923c32d1a0c0c2ce6b1e2 for a test run tonight. If all goes well I will come up with fixups for running this solely over the weekends.

nfeske, 2020-04-08:
> may you please add the commits

Sure! I merged them just now.

alex-ab, 2020-04-09:
The test runs well and with some manual tweaking the results are available on our ci webpage. Please add e610d7b7bba93210a5c3de30a03f359d8e430e12 to restrict the tests to run on weekends only.

chelmuth, 2020-04-09:
Merged to staging.

chelmuth, 2020-04-17:
The commits entered master. May we close this issue?

alex-ab, 2020-04-20:
Yes.

closed 2020-04-20 by alex-ab