genodelabs/genode #3688: qt5: handle Genode signals as Qt signals in QPA plugin

cproc, 2020-03-10:
Handling the Genode signals directly caused problems with multi-threaded applications (for example, page faults when resizing the Arora window) and can also lead to nested signal handler execution when blocking via `Libc::suspend()`, so it's still necessary to handle the Genode signals as Qt signals.

Related to #3644

nfeske, 2020-03-10:
Thank you for the super quick investigation, @cproc!

chelmuth, 2020-03-11:
For completeness: This solution just minimizes the mutex-contention window to the emission of queued `Q_SIGNALS` connections. A real fix had to move the nitpicker-signal handling (which has to be I/O level according to @cproc) behind the VFS border so that Libc synchronization, suspend, and also monitor will not end up in double suspend or unexpected nested execution of application/library code.

commit 04b667c4 (2020-03-11): qt5: handle Genode signals as Qt signals in QPA plugin

Fixes #3688
commit 85356886 (2020-03-26): qt5: handle Genode signals as Qt signals in QPA plugin

Fixes #3688
closed 2020-03-26 by nfeske