genodelabs/genode #3691: base-linux: re enable build for platforms other than x86

trimpim, 2020-03-12:
With the `20.02` release it is no longer possible to build `base-linux` on platforms other than `x86`. I fixed this in my branch.

This seems to be collateral damage caused by #2829.

@jklmnn may you have time to look at these patches?

jklmnn, 2020-03-12:
I can take a look at it although it may take some time.

jklmnn, 2020-03-17:
I've taken a look into your patches. I think a better approach would be to only allow the combination `KERNEL=linux BOARD=pc` (which triggers base-linux-hw) only on x86 and have all other platforms only accept `KERNEL=linux BOARD=linux`. Switching the board should already select the correct files.
If you want to have driver support for base-linux on ARM this is a totally different story as this has never been tested and there are some road blocks for sure (e.g. the current kernel image is x86 only).

trimpim, 2020-03-18:
@jklmnn I will change the patch, but this also might take some time.

trimpim, 2020-04-06:
Upon further thoughts about this issue I'm no longer sure how to proceed.

I have different build errors on arm_v7a and armhf_v7a.
 - on `arm_v7a` building fails because `SYS_mmap2` is not declared in `repos/base-linux/src/lib/syscall/linux_syscalls.h:206` (this didn'd change since release 19.11)
 - on `armhf_v7a` the changes in #2829 caused three build errors that  are addressed by my patches in this issue.

We have patches that enable base-linux on armhf_v7a for a specific hardware (requiring a target specific cross toolchain to build the Linux specific parts).

@nfeske what would you suggest I should do with this issue and my patches?

chelmuth, 2020-04-07:
> Upon further thoughts about this issue I'm no longer sure how to proceed.
>
> I have different build errors on arm_v7a and armhf_v7a.
>
>     * on `arm_v7a` building fails because `SYS_mmap2` is not [...]

Do I understand correctly that base-linux no longer builds for your platform using the standard tool chain despite you didn't change anything? AFAIK `SYS_mmap2` should be available on all platforms according to https://elixir.bootlin.com/linux/v4.20.17/source/include/uapi/asm-generic/unistd.h#L613 This is the first issue that should be addressed independently of `armhf_v7a`.

>     * on `armhf_v7a` the changes in #2829 caused three build errors that  are addressed by my patches in this issue.

I'm fine with merging 2826c18ba297b6f12f05001e4b778c43cbd686e1 as well as 89fd936ff96540e098070b2497bc4c887f5ede15, but cf0e9b52b49f96866c07bcd8102d79a991239dfc needs refactoring to remove the excessive use of the shun `#ifdef...#endif` pattern. Please move platform-specific parts into `spec`d header files.

trimpim, 2020-04-07:
> Do I understand correctly that base-linux no longer builds for your platform using the standard tool chain despite you didn't change anything?

I forgot to mention in the initial description that I could only verify this on `armhf_v7a`. I was never able to build base-linux for `arm_v7a` (releases 19.08, 19.11 and 20.02).

> I'm fine with merging 2826c18 as well as 89fd936, but cf0e9b5 needs refactoring to remove the excessive use of the shun #ifdef...#endif pattern. Please move platform-specific parts into `spec`d header files.

From the perspective that Genode could only test this with an additional patch that enables `armhf_v7a`, I see two possible ways to proceed;
  - I update cf0e9b5 to use the `spec` mechanism and Genode integrates the patches 
  - I keep them in my repository and maintain them for my self

Which ever fits you the best, I am happy with it.

nfeske, 2020-04-07:
Currently base-linux undergoes no automated tests for ARM at Genode Labs. Under this circumstances, the most pragmatic approach would be to maintain the patches locally to your repository because we don't detect regressions like this reliably.

Should we go forward with extending our automated test towards Linux on ARM and thereby make this platform support official, your patches will become a handy reference for fixing the outstanding issues.

Would that approach be ok for you, @trimpim?

trimpim, 2020-04-07:
@nfeske yes, this is ok for me.

closed 2020-04-07 by trimpim