genodelabs/genode #3699: x86/platform_drv: support MSI-X

alex-ab, 2020-03-16:
Parsing for MSI-X cap and setup of the MSI tables is missing in the driver currently.

alex-ab, 2020-03-25:
I now could test on 2 machines the MSI-X feature and make the commit ready for review. 

The platform driver report now contains also the PCI caps, where MSI-X is 0x11. Attached a log from a successful detection by the NVMe driver.

```
init -> acpi_report_rom] report 'platform_drv -> pci'
[init -> acpi_report_rom]   <pci>
[init -> acpi_report_rom]   	<device bdf="0:0.0" vendor_id="0x8086" device_id="0xc00" class_code="0x60000" bridge="no" cap="0x9"/>
[init -> acpi_report_rom]   	<device bdf="0:1.0" vendor_id="0x8086" device_id="0xc01" class_code="0x60400" bridge="yes" cap="0xd" cap="0x1" cap="0x5" cap="0x10"/>
[init -> acpi_report_rom]   	<device bdf="0:2.0" vendor_id="0x8086" device_id="0x412" class_code="0x30000" bridge="no" cap="0x5" cap="0x1" cap="0x13"/>
[init -> acpi_report_rom]   	<device bdf="0:14.0" vendor_id="0x8086" device_id="0x8c31" class_code="0xc0330" bridge="no" cap="0x1" cap="0x5"/>
[init -> acpi_report_rom]   	<device bdf="0:16.0" vendor_id="0x8086" device_id="0x8c3a" class_code="0x78000" bridge="no" cap="0x1" cap="0x5"/>
[init -> acpi_report_rom]   	<device bdf="0:19.0" vendor_id="0x8086" device_id="0x153a" class_code="0x20000" bridge="no" cap="0x1" cap="0x5" cap="0x13"/>
[init -> acpi_report_rom]   	<device bdf="0:1b.0" vendor_id="0x8086" device_id="0x8c20" class_code="0x40300" bridge="no" cap="0x1" cap="0x5" cap="0x10"/>
[init -> acpi_report_rom]   	<device bdf="0:1c.0" vendor_id="0x8086" device_id="0x8c10" class_code="0x60400" bridge="yes" cap="0x10" cap="0x5" cap="0xd" cap="0x1"/>
[init -> acpi_report_rom]   	<device bdf="0:1d.0" vendor_id="0x8086" device_id="0x8c26" class_code="0xc0320" bridge="no" cap="0x1" cap="0xa" cap="0x13"/>
[init -> acpi_report_rom]   	<device bdf="0:1f.0" vendor_id="0x8086" device_id="0x8c4e" class_code="0x60100" bridge="no" cap="0x9"/>
[init -> acpi_report_rom]   	<device bdf="0:1f.2" vendor_id="0x8086" device_id="0x8c02" class_code="0x10601" bridge="no" cap="0x5" cap="0x1" cap="0x12"/>
[init -> acpi_report_rom]   	<device bdf="0:1f.3" vendor_id="0x8086" device_id="0x8c22" class_code="0xc0500" bridge="no"/>
[init -> acpi_report_rom]   	<device bdf="1:0.0" vendor_id="0x1987" device_id="0x5007" class_code="0x10802" bridge="no" cap="0x1" cap="0x5" cap="0x10" cap="0x11"/>
[init -> acpi_report_rom]   	<device bdf="2:0.0" vendor_id="0x8086" device_id="0x88e" class_code="0x28000" bridge="no" cap="0x1" cap="0x5" cap="0x10"/>
[init -> acpi_report_rom]   </pci>
[init -> platform_drv] nvme_drv -> : assignment of PCI device 01:00.0 succeeded
[init -> platform_drv] 01:00.0 adjust IRQ as reported by ACPI: 11 -> 16
[init -> platform_drv] 01:00.0 uses MSI-X vector 0x7f, address 0xfee00018
[init -> nvme_drv] NVMe PCIe controller found (0x1987:0x5007)
...
```

nfeske, 2020-03-26:
Would you mind removing the commented-out code (`#if 0`) from the patch?

alex-ab, 2020-03-26:
Sure, thanks for the pointer. Added an updated commit.

nfeske, 2020-03-26:
Thanks. I merged your commit to staging now.

chelmuth, 2020-03-27:
@alex-ab please note that the patch breaks `tool/depot/create genodelabs/bin/x86_64/platform_drv`

```
In file included from /data/depot/genodelabs/src/platform_drv/2020-03-26/src/drivers/platform/spec/x86/pci_session_component.h:39,
                 from /data/depot/genodelabs/src/platform_drv/2020-03-26/src/drivers/platform/spec/x86/irq.cc:21:
/data/depot/genodelabs/src/platform_drv/2020-03-26/src/drivers/platform/spec/x86/pci_device_component.h: In member function ‘unsigned int Platform::Device_component::_configure_irq(unsigned 
/data/depot/genodelabs/src/platform_drv/2020-03-26/src/drivers/platform/
```

chelmuth, 2020-03-27:
FIxup was easy, see 7b3785c16682e26ec5d9966c012adf11e1e8be55.

commit 79c73d59 (2020-03-26): x86: support msi-x by platform driver

Fixes #3699
closed 2020-04-17 by chelmuth