genodelabs/genode #3702: os: use Request_stream API in NVMe driver

cnuke, 2020-03-19:


cnuke, 2020-03-19:
Commit 11fcffc contains the change. I took the oppertunity to refactor some parts of the driver in addition to the API modification, which leads to the rather large diff.

So far the `nvme.run` tests works well on Qemu as well as on hardware. Yesterday I took the driver for a spin running in Sculpt on my DS57U test machine without any problems, today I deployed on my T470p.

@ssumpf I would appreciate if you could take a look and review the changes, I'll defer adding the fixed label until then.

ssumpf, 2020-03-19:
@cnuke: LvGTM and the new DMA handling seems to make things easier.
Since you are at it, maybe we could take care of this also:
```
/*
 * XXX For now let the device decide if it wants to deallocate
 *     the blocks or not.
 *
 * b.write<Nvme::Sqe_io::Cdw12::Deac>(1);
 */
```

cnuke, 2020-03-19:
@ssumpf yeah - the reason we I don't set the bit unconditionally is because AFAICT its not part of the NVMe 1.2 spec and I do not how such devices will cobe with that. At the moment the driver does not care about the version.

So the comment is there as a reminder that we could set the bit in the future. After all I intepret the spec that is merely a hint on how the device should behave.

ssumpf, 2020-03-20:
@cnuke: Okay, thanks for the explanation. So, everything is fine from my side.

cnuke, 2020-03-22:
Alright, thanks for taking a look.

nfeske, 2020-03-26:
Merged to staging.

cnuke, 2020-03-30:
@chelmuth please merge da60486 to staging.

chelmuth, 2020-03-30:
Thanks, merged to staging.

commit 6dbfe185 (2020-03-26): os: use Request_stream API in NVMe driver

As a result of the API change the memory handling could be simplified.
Since the Block session dataspace is now directly used for DMA, we
actually only have to provide the memory for setting up PRP lists for
large requests (for the moment more than 8 KiB of data).

As we limit the maximum data transfer length to 2 MiB, we get by with
just a page per request. Those memory is allocated beforehand for the
maximum number of I/O requests, which got bumbed to 512 entries. Since
not all NVMe controllers support such large a maximum data transfer
length and t [... cut]
commit 150d1437 (2020-04-17): os: use Request_stream API in NVMe driver

As a result of the API change the memory handling could be simplified.
Since the Block session dataspace is now directly used for DMA, we
actually only have to provide the memory for setting up PRP lists for
large requests (for the moment more than 8 KiB of data).

As we limit the maximum data transfer length to 2 MiB, we get by with
just a page per request. Those memory is allocated beforehand for the
maximum number of I/O requests, which got bumbed to 512 entries. Since
not all NVMe controllers support such large a maximum data transfer
length and t [... cut]
closed 2020-04-17 by chelmuth