genodelabs/genode #3703: hw: abort in core during `Hw_native_pd::upgrade_cap_slab`

nfeske, 2020-03-19:
The `upgrade_cap_slab` RPC function may trigger an `Out_of_caps` exception, which is not reflected to the client.

skalk, 2020-03-19:
Ouch, thank you for the fix!

nfeske, 2020-03-19:
This problem triggered during my work on the tool_chain_auto scenario. @skalk could you please have a look at my patch? I fixed the abort but I am not certain that the `Out_of_caps` exception is plausible in the first place.

skalk, 2020-03-19:
@nfeske: the upgrade_cap_slab method uses the sliced_heap of the pd_session to allocate a new slab block, and adds that to the Tslab, which is used for capability allocation of that PD. I assume that the allocation of a new ram block with the sliced_heap can trigger the 'Genode::Quota_guard<Genode::Cap_quota>::Limit_exceeded', doesn't it?

nfeske, 2020-03-20:
@skalk that's the answer I just wanted to hear. Yes, this makes perfect sense.

nfeske, 2020-03-23:
With commit https://github.com/genodelabs/genode/commit/88b643979ebcec34076d60588e9bf2b1110fad9e, the tool_chain_auto.run script runs successfully on base-hw now.

The commit solves a deadlock situation that could occur whenever the upgrade condition triggers during an RPC towards the parent, e.g., a session request. In this situation, the attempt to call `Env::upgrade` as a side effect of the IPC mechanism would attempt to grab the `Env::_lock` another time. The commit fixes this circular dependency by directly issuing a resource request to the parent 
instead of relying on the corner-case handling of `Expanding_parent_client::upgrade`.

commit d5518aa5 (2020-03-19): base-hw: let upgrade_cap_slab reflect Out_of_caps

This patch reflects the Out_of_caps exception from core to the client,
in addition to the already covered Out_of_ram exception. It thereby
eliminates a potential abort in core, which I observed with the
tool_chain_auto.run script:

  Kernel: RPC upgrade_cap_slab
  Error: Uncaught exception of type 'Genode::Quota_guard<Genode::Cap_quota>::Limit_exceeded'
  Warning: abort called - thread: entrypoint

In addition to propagating the exception, the patch add the client-side
exception-handling code to the base library.

Fixes #3703
closed 2020-03-26 by nfeske