genodelabs/genode #3704: base: wait_and_dispatch_one_io_signal may stall

skalk, 2020-03-19:
There is a race-condition in the interplay of the thread, which delivers signals to an 'ep' thread via IPC. The signal deliverer thread uses _process_incoming_signals therefore, which calls the 'ep' thread here:
https://github.com/genodelabs/genode/blob/bf36d9eb480f311dc08ff16aa8756529fe725d4c/repos/base/src/lib/base/entrypoint.cc#L134
But the 'ep' thread unconditionally enters the signal reception code in _wait_and_dispatch_one_io_signal by ignoring the already set _signal_recipient, see here:
https://github.com/genodelabs/genode/blob/bf36d9eb480f311dc08ff16aa8756529fe725d4c/repos/base/src/lib/base/entrypoint.cc#L203
Thereby, it may enter:
https://github.com/genodelabs/genode/blob/bf36d9eb480f311dc08ff16aa8756529fe725d4c/repos/base/src/lib/base/entrypoint.cc#L227
and will block forever, if no other signal arrives, although there is still the signal inflight, which the signal deliver thread tries to send via IPC. This happens nowadays regularily on base-hw when executing the test-fs_rom_update_ram test on x86_64 in the depot_autopilot run-script.

chelmuth, 2020-03-20:
Thanks for recording this issue. I'm currently working on a stable test case that triggers exactly the wrong behavior. Then, I'm going to implement the solution @nfeske and me drafted this morning based on your and my own investigations.

chelmuth, 2020-03-23:
The following commits fix the issue for my tests.
45d20dc3ba base: rework interplay of signal-proxy and entrypoint
276cd790d5 depot_autopilot: test case for entrypoint

nfeske, 2020-03-24:
The retry loop made me wonder a bit. By default, the retry utility makes ~0U attempts. Is this your intention.

Thanks to your patch, the mechanism looks so much cleaner now - very cool!

Now I'm very curious how the automated tests fare. :-)

nfeske, 2020-03-24:
> Is this your intention.

I guess this is not a problem because the blocking-canceled condition happens only sporadically and does not occur repeatedly.

skalk, 2020-03-24:
@chelmuth with your commits, the issue I could safely reproduce with hw/x86 on Qemu beforehand does not occur anymore! Respect for the sane solution in this tricky part.

chelmuth, 2020-03-24:
> The retry loop made me wonder a bit. By default, the retry utility makes ~0U attempts. Is this your intention.

The retry loop was already part of the implementation before my patch.

https://github.com/genodelabs/genode/blob/cfb49c73165ee399d2ab116011d5918b8e96d03a/repos/base/src/lib/base/entrypoint.cc#L132-L136

IIRC it is needed because of canceled blocking (at least on NOVA) in certain cases. I think @alex-ab may know more as we briefly talked about it in context of

https://github.com/genodelabs/genode/blob/cfb49c73165ee399d2ab116011d5918b8e96d03a/repos/base/src/lib/base/mutex.cc#L20-L30

alex-ab, 2020-03-24:
I have no further information about this specific retry<Blocking_cancel> loop. I looked back a bit in history, and the retry loop got introduced by 051e84c4b47f89b0d321147ec01cd0cee4e0961c commit and was moved around several times.

alex-ab, 2020-03-24:
Looking at 051e84c the original code code changed from 

```
		/*
		 * It might happen that we try to forward a signal to the entrypoint,
		 * while the context of that signal is already destroyed. In that case
		 * we will get an ipc error exception as result, which has to be caught.
		 */
		try {
			constructor_cap.call<Server::Constructor::Rpc_signal>();
		} catch(Genode::Ipc_error) { }
```
to

```
			_sig_rec->block_for_signal();

			/*
			 * It might happen that we try to forward a signal to the
			 * entrypoint, while the context of that signal is already
			 * destroyed. In that case we will get an ipc error exception
			 * as result, which has to be caught.
			 */
			retry<Genode::Blocking_canceled>(
				[&] () { _signal_proxy_cap.call<Signal_proxy::Rpc_signal>(); },
				[]  () { PWRN("Catched Blocking_canceled on signal processing"); }
			);
```

The description still talks about Ipc error catching, as in the original part - this would also make sense. Why now the Blocking_canceled retry loop got introduced @nfeske may know.

nfeske, 2020-03-24:
Thanks for the forensics.

> Why now the Blocking_canceled retry loop got introduced @nfeske may know.

I frankly cannot remember. Admittedly, the comment does not really match the code anymore. Nowadays, the blocking-canceled exception comes into effect only during thread destruction. In the case at hand, when the signal proxy gets destructed. Since this rarely happens with today's components, this condition probably never occurs in practice.

Maybe grep'ping the autopilot for the warning message would bring us on the right track?

Anyway, as we are going to eventually remove the cancel-blocking mechanism in the not-too-distant future, we can remove the retry loop along with it then.

nfeske, 2020-03-26:
Merged to staging.

commit 2c461e84 (2020-03-26): base: rework interplay of signal-proxy and entrypoint

The former scheme left open a race window between
_process_incoming_signals() and wait_and_dispatch_one_io_signal()
resulting in both threads calling block_for_signal() and blocking
forever with one unprocessed signal.

Fixes #3704
commit e63c5e6c (2020-04-17): base: rework interplay of signal-proxy and entrypoint

The former scheme left open a race window between
_process_incoming_signals() and wait_and_dispatch_one_io_signal()
resulting in both threads calling block_for_signal() and blocking
forever with one unprocessed signal.

Fixes #3704
closed 2020-04-17 by chelmuth