genodelabs/genode #3705: part_block: partitions with index greater than 32 unavailable

ehmry, 2020-03-20:


ehmry, 2020-03-20:
Fixed at 1b8f44e80801bbb8d3dba813b72eed4c6a5a5ca6.

ehmry, 2020-03-20:
FWIW the problematic line is here: https://github.com/ehmry/genode/blob/1b8f44e80801bbb8d3dba813b72eed4c6a5a5ca6/repos/os/src/server/part_block/main.cc#L337

ssumpf, 2020-03-20:
@ehmry: I deliberately limited the number of sessions in order to make `_wakeup_clients` faster, but that may be premature optimization. @chelmuth: I am fine with the supposed change for my part.

ehmry, 2020-03-20:
Does this mean I have to do a benchmark now? :P

ssumpf, 2020-03-20:
@ehmry: Worse, two benchmarks (old and new) ;)

ehmry, 2020-03-21:
`Main::_wakeup_clients` is on the hot path but changing the size of the array doesn't have a noticeable effect.

ssumpf, 2020-03-21:
@ehmry: I know and was just kidding.

commit 8ecce039 (2020-03-23): part_block: support for up to 128 session

Allow sessions for each of the 128 sessions supported by GPT.

Fix #3705
closed 2020-03-26 by nfeske