genodelabs/genode #3707: libc: setting rtc to year 2100 blocks application startup on 32-bit

nfeske, 2020-03-20:
If changing the rtc setting in ports/run/bash.run from
```
<inline name="rtc">2018-01-01 00:01</inline>
```
to
```
<inline name="rtc">2100-01-01 00:01</inline>
```
bash mysteriously fails to start up, but only on 32-bit platforms. Smells like an integer overflow.

alex-ab, 2020-03-20:
Just a guess - https://en.wikipedia.org/wiki/Year_2038_problem

nfeske, 2023-12-07:
> Nevertheless, the diff of the mentioned for-loop looks pretty suspicious and could already fix the infinite-loop issue:

Indeed, with the patch it works. Wow!

Thanks a lot for the impressive forensic achievement.

Would you mind adding the patch to repos/libports/src/lib/libc/patches/? (or me doing this?)

nfeske, 2023-12-07:
Splendid! Merged to staging.

nfeske, 2023-12-13:
The fix entered the master branch today.

closed 2023-12-13 by nfeske