genodelabs/genode #3712: VFS ram_fs plugin: add support for write limiiting

trimpim, 2020-03-24:


trimpim, 2020-03-25:
I have added the limit support to the ram_fs plugin in bd0ed72  and added a test to `test-libc_vfs` in  f20af97.

In 076058e I have tried to create a stand alone test for the feature that doesn't use libc. but this is not finished yet, as it fails to write. I will further investigate this in the next days.

nfeske, 2020-03-26:
Thanks for taking the issue from the mailing list to here.

The proper implementation of this feature would need to account the gross costs, not the net costs (as done in your patch) though. Consider the following:

* The allocation of meta data for file and directory nodes.
* The allocation of page-table structures for the ram-fs-chunk data structure.
* The meta-data overhead of the used allocator.
* An untrusted client can still exhaust the RAM of the VFS server by creating an infinite amount of directories.
* Similarly, a malicious client could exploit the fact that memory allocations for the file's backing store happen at a coarse granularity. E.g., by writing one byte at every 4K boundary of a file, the gross costs in the VFS server are 4096 times the accounted net costs.

To account for these indirect costs, counting the bytes written won't suffice. To catch all these cases, the ram vfs-plugin instance should better work on a dedicated heap instance that obtains its RAM dataspaces through a `Constrained_ram_allocator` (base/ram_allocator.h). However, this introduces additional problems. In particular, all code within the ram fs must become able to properly cope with `Out_of_ram` and `Out_of_caps` exceptions since it can no longer rely on the implicit use of the resource-request mechanism. I'm afraid that this change requires a more invasive redesign, which would affect not only the vfs but also the ram_fs server. Since I'm planning to retire the latter (because its feature set is completely covered by the vfs server), I would defer such a change until the ram_fs server is gone.

While looking at your patch, I also noticed the following minor points that are worth sharing:

* The `Number_of_bytes` utility is solely meant for parsing purposes, not as a type for member variables.
* The limit is defined at construction time and never changes. Hence, it should be stored in a const member.
* Our coding style doesn't use C++-style comments.
* The warning about a potentially misconfigured limit is prone to false-positives and false-negatives. When considering a vfs configuration as follows, no warning would be given:
  ```
  <resource type="RAM" quantum="10M"/>
  <config>
   <vfs>
    <dir name="a"> <ram limit="8M"/> </dir>
    <dir name="b"> <ram limit="8M"/> </dir>
   </vfs>
  </config>
  ```
  In cases where the RAM of the VFS server is upgraded on demand (think of the RAM fs in Sculpt), the definition of a limit would always yield a warning.
  Given these uncertainties, I argue that sanity checks like this should better be covered by offline integration tools.

Please don't feel urged to address those points though. As I have written in https://github.com/genodelabs/genode/issues/3655#issuecomment-604384465 a proper solution is often not immediately attainable. The issue nicely documents the requirement along with a fix for your concrete use case. I will definitely pick up the topic once my time permits.