genodelabs/genode #3715: test-ldso depot-autopilot test fails sporadically on imx53_qsb

chelmuth, 2020-03-27:
The test sporadically misses the following output line (marked with *).

```
 [init -> depot_autopilot] 0.975 [init -> test-ldso] Shared-object API
 [init -> depot_autopilot] 0.981 [init -> test-ldso] -----------------
*[init -> depot_autopilot] 0.993 [init -> test-ldso] Global object constructed
 [init -> depot_autopilot] 1.018 [init -> test-ldso] 
 [init -> depot_autopilot] 1.024 [init -> test-ldso] Destruction
 [init -> depot_autopilot] 1.030 [init -> test-ldso] -----------
```

I traced this back to an implausible pointer value in the following source-code line when loading the shared object _test-ldso_lib_dl.lib.so_.
https://github.com/genodelabs/genode/blob/46fa8197a21289f54bb96c814bf62fc14d149c6e/repos/base/src/lib/ldso/include/dynamic.h#L160
The pointer is calculated via addition of a _value_ read from the _.dynamic_ section of the newly mapped shared object and the mapping base. Sporadically the _value_ does not correspond to the contents of the shared-object binary. Most interestingly the wrong value stays the same over several test runs.

```
Kernel: DT_HASH _hash_table pre=0x0 post=0x6575b550 d=0x100e0b0/0x6474e550
```

For testing I used the following extensive command line which always builds _ld.lib.so_ and _test-ldso_ and puts all binaries into boot modules.

```
make -C build/arm_v7a run/depot_autopilot TEST_SRCS='' TEST_PKGS=test-ldso K=hw B=imx53_qsb TEST_REPEAT=until_failed TEST_BUILDS="lib/ld test/ldso" TEST_MODULES="ld.lib.so test-ldso test-ldso_lib_1.lib.so test-ldso_lib_2.lib.so test-ldso_lib_dl.lib.so"
```

chelmuth, 2020-03-27:
I tracked the error back to `load_segment_rw()` and discovered that the temporarily mapped boot module does not have the expected content: There has to the first dynamic entry at offset `0x88` with tag `0x1` but in the error case its always `0x15c`. Commit 475de1060b97410fb5c7d958fac33ed1534bf42f adds debug output and also retries the attachment several times. In my test cases, the retry sometimes also produces the wrong value at the same mapping location but always succeeds at a new location. Could this be a caching/TLB issue?

chelmuth, 2020-03-27:
When I dump the first 64 bytes of the mapped ROM, I can asses that the first 64 bytes of the  _test-ldso_lib_dl.lib.so_ ELF binary are accessed as if the `offset=0x2000` parameter of the attachment was just ignored.

chelmuth, 2020-04-03:
The following patch by @skalk looks fairly promising - 3 test runs of 10 mins each without errors so far.

```patch
--- a/repos/base-hw/src/include/hw/spec/arm/page_table.h
+++ b/repos/base-hw/src/include/hw/spec/arm/page_table.h
@@ -580,15 +580,22 @@ class Hw::Page_table
 
                                                addr_t const pt_vo = vo - Section::Pa::masked(vo);
                                                pt.remove_translation(pt_vo, Genode::min(size, end-vo));
+                                               _translation_added((addr_t)&pt, sizeof(Page_table_level_2));
 
                                                if (pt.empty()) {
                                                        Descriptor::invalidate(_entries[i]);
+                                                       _translation_added((addr_t)&_entries[i], sizeof(Section));
                                                        alloc.destruct<Pt>(pt);
                                                }
                                                break;
                                        }
 
-                               default: Descriptor::invalidate(_entries[i]); }
+                               default:
+                                       {
+                                               Descriptor::invalidate(_entries[i]);
+                                               _translation_added((addr_t)&_entries[i], sizeof(Section));
+                                       }
+                               }
 
                                /* check whether we wrap */
                                if (end < vo) return;
```

Now I need to develop a test case (e.g., in test/rm_fault) stressing exactly this part and reproducing the error without the patch.

chelmuth, 2020-04-03:
The rationale behind the patch stems from a note in the _Cortex -A8 Technical Reference Manual, 3.2.31 c2, Translation Table Base Register 0_.

> The processor cannot perform a translation table walk from L1 cache. Therefore, if C is
set to 1, to ensure coherency, you must store translation tables in inner write-through
memory. If you store the translation tables in an inner write-back memory region, you
must clean the appropriate cache entries after modification so that the mechanism for
the hardware translation table walks sees them.

In our case, a former ROM detachment did invalidate the TLB but did not clean the L1 cache. So, a later attachment (which does not touch the translation table yet) and virtual-memory access will experience a TLB miss but _may_ not fault because stale translation-table entry values are used on translation-table walk. This renders demand paging ineffective.

chelmuth, 2020-04-03:
I added the test in f25b6b3d6f which triggers the issue robustly while it no longer occurs with 2cc6545b9e.

skalk, 2020-04-06:
I've reworked the WIP commit to reflect the change in the name of the function too. Moreover, I could see that the cache maintainance was done in bootstrap for rpi1 - although MMU and caches are not even enabled. I removed the maintainance here.

chelmuth, 2020-04-07:
Merged 935adaa748244834ae0a9ec614f92e015721edbc and 0f6ee858515205e67df267e13a5f9c98f00edf2e.

commit 935adaa7 (2020-04-07): hw: cache maintainance on mapping removal too

Older ARM processors like ARMv6, or Cortex A8 need to write back changes
of the page-tables to physical ram because the MMU does not use the cache.
This naturally needs to be done not only when adding a mapping,
but on removal too.

Fix #3715
closed 2020-04-17 by chelmuth