genodelabs/genode #3717: nic_router: make acceptable dhcp offers for uplink configurable

alex-ab, 2020-03-28:
Is such a feature desirable ? I have a usecase where multiple dhcp server are running, serving different purposes - one for pxe boot, one for the uplink/internet. Depending on the set options set by the DHCP offers, I would like to configure to decline to accept some of the offers, e.g. no router ip address set or no dns server ip set.

alex-ab, 2020-03-28:
The commit showcases what I mean. (The configurable is missing, therefore WIP)

m-stein, 2020-03-30:
@alex-ab I'm not sure whether this is a good idea. How about selecting a specific DHCP server by its IP instead through means of configuration (e.g. ``<domain ... dhcp_server="10.0.2.43">``)?. I have the feeling that the solution you suggest "misuses" unrelated indicators for identifying a server. But maybe I got you wrong...

alex-ab, 2020-03-30:
> IP instead through means of configuration (e.g. <domain ... dhcp_server="10.0.2.43">)?
Is this already possible or do you mean it as suggestion ? 

I'm fine with that a human should do the parsing and take the decision. This would require to "see" the dhcp offers in order to be able to take a decision, e.g. if you're running on Sculpt in a unknown network setup.

chelmuth, 2020-03-30:
> Is such a feature desirable?

Operating more than one DHCP server inside one network segment was and still is a balancing act. The first rule of all howtos in this regard is: Make sure the servers serve *distinct* IP ranges (which is in most cases equivalent to MAC addresses ignoring VLANs). Therefore, I don't have a desire to solve this issue on client side.

> I have a usecase where multiple dhcp server are running, serving different purposes - one for pxe boot, one for the uplink/internet. Depending on the set options set by the DHCP offers, I would like [...]

There is still the open question of what will happen with all other hosts in the network. Maybe your answer is: The PXE server only serves MAC addresses of my test hardware. With this, you violate the first rule of multiple DHCP servers. I'd suggest you ban the MAC address of the test hardware from your standard DHCP server and add all required options to your custom server effectively creating two distinct network/MAC address ranges.

If this does not work, you may still go for a NAT router in your LAN shielding your test network from the LAN.

m-stein, 2020-03-30:
@alex-ab This was just a suggestion. I'd prefer the solution of @chelmuth if it works for you.

alex-ab, 2020-03-30:
I see.

closed 2020-03-30 by alex-ab