genodelabs/genode #3718: test-trace_logger: failure due to wrong policy rules

tomga, 2020-03-29:
Important part of logs:
```
[init -> test -> trace_logger] Warning: no policy defined for label 'init -> test -> cpu_burner.2'
[init -> test -> trace_logger] Warning: no policy defined for label 'init -> test -> cpu_burner.1'
[init -> test -> trace_logger] Warning: no policy defined for label 'init -> test -> test-trace_logger'
```
where the policies in `runtime` file are like:
```
init -> dynamic -> test-trace_logger -> cpu_burner.*
```
I assume that something changed during last year in component naming and updating rules is a proper fix.

tomga, 2020-03-29:
9365c395b9481d24b8af000edb3c417f35ddea98 the test for me on linux, rpi and my rpi3_32.

chelmuth, 2020-03-30:
The reason for these rules is that the test was developed for `run/depot_autopilot` where the labels differ from your log. Am I right in assuming that you're using `run/test PKG=test-trace_logger`?

tomga, 2020-03-30:
Yes, I do. Is this method considered deprecated? Is there a way to make this test work in both cases?

tomga, 2020-03-30:
And one more thing that I think does not deserve issue: a1565c67bb00a7225f9e9b3a9cc89aa4430cbf2f
I found that it is wrong in Genode Foundations. I think I pointed it to a right file now.

chelmuth, 2020-03-30:
> Yes, I do. Is this method considered deprecated? Is there a way to make this test work in both cases?

This is a current shortcoming of _test.run_ as it does not replicate the depot_autopilot scenario very well. For the time being you may just use `run/depot_autopilot TEST_PKGS=test-trace_logger`.

tomga, 2020-03-30:
Ok, thanks.

closed 2020-03-30 by tomga