genodelabs/genode #3720: libc: provide C++ runtime symbols in ABI

chelmuth, 2020-03-31:
C++ components that just depend on libc/posix but do not use the STL (stdcxx) may have dependencies to the C++ runtime, e.g., when using exceptions. These symbols are always provided by ld.lib.so which contains the basic C++ support. With this assumption it is safe to add the C++ runtime support symbols to the libc ABI to resolve potential dependencies at link time and resolve the symbol lookup in ld.lib.so at runtime. This approach keeps the components base ABI agnostic.

chelmuth, 2020-04-02:
I merged 38c0e9dadd4de3527ab27a1637737518dcff53b1 already to get rid of the nightly error in stdin2out. The cleanup work is still in progress currently stalling at the ldso-startup knot.

chelmuth, 2020-04-02:
In 17db4e31bb0be2a586bafd6d538a2579426ac1ab I did what was feasible to entangle the inter-dependencies of the former _ldso-startup_ static library (now named _ldso_so_support_ because it also provides exidx hooks for shared objects on ARM), the lib/prg targets, and the API archives.

The approach is not 100% satisfying as _ldso_so_support_ is now archived in base, so, and libc api archives. The dependency chain is as follows.

- The static library _ldso_so_support_ must be linked to all shared objects/libraries as it provided local hooks for the linker to execute constructors and also lookup information for ARM exidx handling.
- The dependency above is expressed by _dep_lib.mk_ adding _ldso_so_support_.lib as dependency of any shared library in _var/libdeps_.  <-- I could not replace this solution by any other that does not render any target transitively depending on the static library despite it only uses the abi.so files.
- Because of the dependency added above the build of prg and lib targets using any shared library needs _ldso_so_support.mk_ at build time. <-- Here I failed to find a solution that robustly dummies _ldso_so_support.mk_ in cases where it is not needed actually.

Altogether I had to add _ldso_so_support_ to all APIs that represent a runtime environment: base, libc, and finnaly so. The latter is the only place where it should be required as only src archives containing shared objects actually need the _ldso_so_support_. What already works now is to use either _base_ or _libc_ in component targets as well as in shared libraries.

nfeske, 2020-04-22:
The commit reached the master branch. Can we close this issue?

chelmuth, 2020-04-22:
The base dependency cleanup is still in progress. I'll close the issue with the final commit.

chelmuth, 2020-05-08:
I finally fixed the remaining adaptions in this issue with the following series.

d7a8e6fcd3 libc: fix CC_C_OPT in libm
6532b133ea Remove misleading "pipe" utility and test
631018b811 Cleanup shared-object support mechanics
96ba6dc359 Remove false API dependencies from libc/posix components

I left out Qt5 for now and will put my findings into a dedicated issue.

nfeske, 2020-05-08:
That's so much nicer now! :-)

commit 50287b05 (2020-05-11): Remove false API dependencies from libc/posix components

Fixes #3720
commit c11d9b7b (2020-05-18): Remove false API dependencies from libc/posix components

Fixes #3720
closed 2020-05-18 by nfeske