genodelabs/genode #3723: ieee754 test fails on Qemu/rpi3

chelmuth, 2020-04-06:
The ieee754 test produces unstable results like the diff of two runs below depicts.

```diff
--- 1_
+++ 4_
@@ -11,17 +11,17 @@
 [init -> test1] DBL_MAX_EXP = 1024
 [init -> test1] LDBL_MAX_EXP = 16384
 [init -> test1] 
-[init -> test1] FLT_EPSILON = 1.1921e-07 = 0x1p-23
-[init -> test1] DBL_EPSILON = 2.22045e-16 = 0x1p-52
-[init -> test1] LDBL_EPSILON = 1.92593e-34 = 0x1p-112
-[init -> test1] 
-[init -> test1] FLT_MIN = 1.1755e-38 = 0x1p-126
-[init -> test1] DBL_MIN = 2.22508e-308 = 0x1p-1022
-[init -> test1] LDBL_MIN = 3.36211e-4932 = 0x1p-16382
-[init -> test1] 
-[init -> test1] FLT_MAX = 3.40283e+38 = 0x1.fffffep+127
-[init -> test1] DBL_MAX = 1.7977e+308 = 0x1.fffffffffffffp+1023
-[init -> test1] LDBL_MAX = 1.18974e+4932 = 0x1.ffffffffffffffffffffffffffffp+16383
+[init -> test1] FLT_EPSILON = 1.19209e-07 = 0x1p-23
+[init -> test1] DBL_EPSILON = 2.22044e-16 = 0x1p-52
+[init -> test1] LDBL_EPSILON = 1.92592e-34 = 0x1p-112
+[init -> test1] 
+[init -> test1] FLT_MIN = 1.17549e-38 = 0x1p-126
+[init -> test1] DBL_MIN = 2.22507e-308 = 0x1p-1022
+[init -> test1] LDBL_MIN = 3.3621e-4932 = 0x1p-16382
+[init -> test1] 
+[init -> test1] FLT_MAX = 3.40282e+38 = 0x1.fffffep+127
+[init -> test1] DBL_MAX = 1.79769e+308 = 0x1.fffffffffffffp+1023
+[init -> test1] LDBL_MAX = 1.18973e+4932 = 0x1.ffffffffffffffffffffffffffffp+16383
 [init -> test1] 
 [init -> test1] sizeof(float) = 4
 [init -> test1] sizeof(double) = 8
@@ -30,7 +30,7 @@
 [init -> test1] x + y, with x = 9007199254740994.0 and y = 1.0 - 1/65536.0 (type double).
 [init -> test1] The IEEE-754 result is 9007199254740994 with double precision.
 [init -> test1] The IEEE-754 result is 9007199254740996 with extended precision.
-[init -> test1] The obtained result is 9007199254740996.
+[init -> test1] The obtained result is 9007199254740994.
 [init -> test1] 
 [init -> test1] NAN != NAN --> 1 (should be 1)
 [init -> test1] isnan(NAN) --> 1 (should be 1)
@@ -72,12 +72,12 @@
 [init -> test1] -0 * 0 = -0
 [init -> test1] -0 * -0 = 0
 [init -> test1] Constant expression 1 + DBL_MIN = 1
-[init -> test1] Variable expression 1 + DBL_MIN = 1
+[init -> test1] Variable expression 1 + DBL_MIN = 1.0000000000000002221
 [init -> test1] Constant expression 1 - DBL_MIN = 1
 [init -> test1] Variable expression 1 - DBL_MIN = 1
 [init -> test1] 1/3 in float       : 0x1.555556p-2
-[init -> test1] 1/3 in double      : 0x1.5555555555555p-2
-[init -> test1] 1/3 in long double : 0x1.5555555555555555555555555555p-2
+[init -> test1] 1/3 in double      : 0x1.5555555555556p-2
+[init -> test1] 1/3 in long double : 0x1.5555555555555555555555555556p-2
 [init -> test1] Dec 1.1  = 0x1.199999999999ap+0
 [init -> test1] FLT_MAX  = 0x1.fffffep+127
 [init -> test1] DBL_MAX  = 0x1.fffffffffffffp+1023
@@ -146,7 +146,7 @@
 [init -> test1] pow(0.5, 0) = 1
 [init -> test1] pow(0.5, -0) = 1
 [init -> test1] pow(0.5, 0.5) = 0.707107
-[init -> test1] pow(0.5, -0.5) = 1.41421
+[init -> test1] pow(0.5, -0.5) = 1.41422
 [init -> test1] pow(0.5, 1) = 0.5
 [init -> test1] pow(0.5, -1) = 2
 [init -> test1] pow(0.5, 2) = 0.25
@@ -189,7 +189,7 @@
 [init -> test1] pow(2, -inf) = 0
 [init -> test1] pow(2, 0) = 1
 [init -> test1] pow(2, -0) = 1
-[init -> test1] pow(2, 0.5) = 1.41421
+[init -> test1] pow(2, 0.5) = 1.41422
 [init -> test1] pow(2, -0.5) = 0.707107
 [init -> test1] pow(2, 1) = 2
 [init -> test1] pow(2, -1) = 0.5
```

chelmuth, 2020-04-14:
Several nightly test runs now show that the issue is not related to Qemu but happens also on hardware. Further, it is not related to several test runs but the output produced by the test instances during one run. When comparing the output of the 4 instances there are subtle differences like shown above, for example

```diff
--- /tmp/a	2020-04-14 16:03:43.546613323 +0200
+++ /tmp/b	2020-04-14 16:03:56.463804120 +0200
@@ -12,16 +12,16 @@
 [init -> test.] LDBL_MAX_EXP = 16384
 [init -> test.] 
 [init -> test.] FLT_EPSILON = 1.19209e-07 = 0x1p-23
-[init -> test.] DBL_EPSILON = 2.22044e-16 = 0x1p-52
-[init -> test.] LDBL_EPSILON = 1.92592e-34 = 0x1p-112
+[init -> test.] DBL_EPSILON = 2.22045e-16 = 0x1p-52
+[init -> test.] LDBL_EPSILON = 1.92593e-34 = 0x1p-112
 [init -> test.] 
-[init -> test.] FLT_MIN = 1.1755e-38 = 0x1p-126
-[init -> test.] DBL_MIN = 2.22508e-308 = 0x1p-1022
-[init -> test.] LDBL_MIN = 3.36211e-4932 = 0x1p-16382
-[init -> test.] 
-[init -> test.] FLT_MAX = 3.40283e+38 = 0x1.fffffep+127
-[init -> test.] DBL_MAX = 1.7977e+308 = 0x1.fffffffffffffp+1023
-[init -> test.] LDBL_MAX = 1.18974e+4932 = 0x1.ffffffffffffffffffffffffffffp+16383
+[init -> test.] FLT_MIN = 1.17549e-38 = 0x1p-126
+[init -> test.] DBL_MIN = 2.22507e-308 = 0x1p-1022
+[init -> test.] LDBL_MIN = 3.3621e-4932 = 0x1p-16382
+[init -> test.] 
+[init -> test.] FLT_MAX = 3.40282e+38 = 0x1.fffffep+127
+[init -> test.] DBL_MAX = 1.79769e+308 = 0x1.fffffffffffffp+1023
+[init -> test.] LDBL_MAX = 1.18973e+4932 = 0x1.ffffffffffffffffffffffffffffp+16383
 [init -> test.] 
 [init -> test.] sizeof(float) = 4
 [init -> test.] sizeof(double) = 8
@@ -30,7 +30,7 @@
 [init -> test.] x + y, with x = 9007199254740994.0 and y = 1.0 - 1/65536.0 (type double).
 [init -> test.] The IEEE-754 result is 9007199254740994 with double precision.
 [init -> test.] The IEEE-754 result is 9007199254740996 with extended precision.
-[init -> test.] The obtained result is 9007199254740996.
+[init -> test.] The obtained result is 9007199254740994.
 [init -> test.] 
 [init -> test.] NAN != NAN --> 1 (should be 1)
 [init -> test.] isnan(NAN) --> 1 (should be 1)
@@ -190,7 +190,7 @@
 [init -> test.] pow(2, 0) = 1
 [init -> test.] pow(2, -0) = 1
 [init -> test.] pow(2, 0.5) = 1.41421
-[init -> test.] pow(2, -0.5) = 0.707107
+[init -> test.] pow(2, -0.5) = 0.707106
 [init -> test.] pow(2, 1) = 2
 [init -> test.] pow(2, -1) = 0.5
 [init -> test.] pow(2, 2) = 4
@@ -509,8 +509,8 @@
 [init -> test.] pow(2, -inf) = 0
 [init -> test.] pow(2, 0) = 1
 [init -> test.] pow(2, -0) = 1
-[init -> test.] pow(2, 0.5) = 1.41421
-[init -> test.] pow(2, -0.5) = 0.707106
+[init -> test.] pow(2, 0.5) = 1.41422
+[init -> test.] pow(2, -0.5) = 0.707107
 [init -> test.] pow(2, 1) = 2
 [init -> test.] pow(2, -1) = 0.5
 [init -> test.] pow(2, 2) = 4
@@ -626,7 +626,7 @@
 [init -> test.] pow(0.5, 0) = 1
 [init -> test.] pow(0.5, -0) = 1
 [init -> test.] pow(0.5, 0.5) = 0.707107
-[init -> test.] pow(0.5, -0.5) = 1.41422
+[init -> test.] pow(0.5, -0.5) = 1.41421
 [init -> test.] pow(0.5, 1) = 0.5
 [init -> test.] pow(0.5, -1) = 2
 [init -> test.] pow(0.5, 2) = 0.25
@@ -669,7 +669,7 @@
 [init -> test.] pow(2, -inf) = 0
 [init -> test.] pow(2, 0) = 1
 [init -> test.] pow(2, -0) = 1
-[init -> test.] pow(2, 0.5) = 1.41422
+[init -> test.] pow(2, 0.5) = 1.41421
 [init -> test.] pow(2, -0.5) = 0.707107
 [init -> test.] pow(2, 1) = 2
 [init -> test.] pow(2, -1) = 0.5
```

May the test experience FPU switching issues?

ssumpf, 2021-07-02:
@chelmuth: e0cdc61 fixes the issue on ARM64 base-hw.

commit 13fa473a (2021-07-02): base-hw: save/restore fpcr register for ARM_V8

The floating-point (fpcr) control register is user land accessible and
controls, for example, rounding mode.

fixes #3723 and related to #4213
commit 4b10aa94 (2021-07-28): base-hw: save/restore fpcr register for ARM_V8

The floating-point (fpcr) control register is user land accessible and
controls, for example, rounding mode.

fixes #3723 and related to #4213
closed 2021-07-28 by chelmuth