genodelabs/genode #3724: <propagate exit="…"/> missing from init.xsd

ehmry, 2020-04-07:


nfeske, 2020-04-07:
Can you please share the reasoning behind the following two changes?

- Use a relative path for base_types.xsd
- Allow "label" attributes within `<parent-provides/>`

AFAIK, the latter is not supported by init (respectively the sandbox library).

ehmry, 2020-04-07:
The relative path allows the XSD file to be used with xmllint without supplying a `--path` argument. The second note is about allowing `<service name="…" label="…"/>` nodes within `<parent-provides>` simply for documentation purposes. When generating deeply nested init configurations I find it useful to attach these labels so that this routing information bubbles up.

For example:
```xml
<parent-provides>
  <service name="ROM" label="core_log"/>
  <service name="ROM" label="kernel_log"/>
  <service name="Platform" label="ahci_drv"/>
  …
</parent-provides>
```

chelmuth, 2020-04-07:
> The relative path allows the XSD file to be used with xmllint without supplying a `--path` argument.

Sorry but I'm sure `--path "base/xsd os/xsd"` is what we really want. It is equivalent to `gcc -I base/include -I os/include` which we are used to. If your environment poses further challenges with this I'd like to hear about them.

> The second note is about allowing `<service name="…" label="…"/>` nodes within `<parent-provides>` simply for documentation purposes. When generating deeply nested init configurations I find it useful to attach these labels so that this routing information bubbles up.

Readability is a valid concern and I want to support it as much as possible. But in my opinion, the addition of ineffective attributes does not improve the readability of configurations. Quite the contrary is the case: These `label` attributes suggest the wrong assumption that service requests to the parent are limited to the labels noted as with labels in policies. A much better alternative is to use XML comments for this purpose.

ehmry, 2020-04-07:
A patch with exit only at 332f8fc9cb9d93cba7c213d4c28cd06f826714db.

nfeske, 2020-04-08:
Thanks @ehmry for the new patch, which I merged to staging just now.

commit 50f511af (2020-04-08): Add <exit propagate="…"/> to Init config schema

Fix #3724
closed 2020-04-17 by chelmuth