genodelabs/genode #3725: Retire Exynos 5 support

skalk, 2020-04-07:
Analogue to issue #2850, and the movement of OMAP4 support into genode world, I'd like to move the support of Exynos 5 to genode world too. The reason behind it is that nobody we are aware of still uses this SoC in combination with Genode. A renewal of the Exynos work to a more recent model is not probable as well due to the missing, or poor documentation. On the other hand, the capacity to test the whole testsuite on real hardware targets is limited at Genode Labs, and therefore well spent on active targets. But we do not want to deliver untested code in the main repository. Therefore, a good compromise seems to be: moving the Exynos port to genode world as well. Thereby, interested parties still can access it, and it is proved to compile, but nothing more.
In contrast to Exynos 5, which runs at least on base-hw and Fiasco.OC, the Exynos 4 line is supported by Fiasco.OC only. I'd like to drop it here, as it seems too much burden to even carry outdated platforms of 3rd party kernels in genode world as well, and the members of Universidad Central Marta Abreu, who initially brought it in, does not seem to work with Genode or that platform anymore.

nfeske, 2020-04-07:
Thanks for presenting such a complete picture. I agree with everything you said and very much appreciate your cleanup work.

chelmuth, 2020-04-17:
Merged to staging.

nfeske, 2020-04-21:
I'm not entirely sure if related to this issue but there is a package-build issue with staging now:

```
In file included from /data/depot/genodelabs/src/base-hw-panda/2020-04-20//src/bootstrap/init.cc:15:
/data/depot/genodelabs/src/base-hw-panda/2020-04-20//src/bootstrap/platform.h:22:10: fatal error: board.h: No such file or directory
 #include <board.h>
```

skalk, 2020-04-22:
@nfeske: Actually it is not this issue, nor this repository ;-). To make the confusion complete: I've introduced the buggy panda package in the commit fixing issue 202 of genode-world (Move OMAP4), but fixed it implicitely in issue commit 203 (Move Exynos5). Sorry, for that not-so-clean practice. I'll sort out the fixup from the second commit, and provide it separately.

commit 0e49336b (2020-04-17): Retire Exynos 5 support (fix #3725)
closed 2020-04-24 by nfeske