genodelabs/genode #3726: base-hw: S2-translation changes do not properly update cache and TLB

zeehha, 2020-04-07:
Hi,
for my thesis-project, I extensively make use of the dataspace attachment/detachment API. I discovered that there are two issues when updating the stage-2 translation for a VM:

-  when attaching a dataspace to a VM, the updates made to the S2-translation do not reliably reach the point-of-unification. This causes the VM to use the old S2-translations.
- when detaching a dataspace of a VM, its TLB entries are not invalidated. Hence, VM accesses to the detached dataspace still remain valid.

Best regards and stay safe,
Chris

skalk, 2020-04-08:
@zeehha thanks for testing and the patch! Indeed, by now the VM session interface for the ARM case was never used with dynamically changing guest-physical memory. I'll have a look at your patch as soon as possible.

zeehha, 2020-04-13:
I discovered, that I missed to update the cache at one point in the page-table. I'm providing an update to the pull-request with the complete fix tomorrow.

Best,
Chris

skalk, 2020-04-22:
@zeehha I finally found the time to review your contribution, thank you for it again.

Unfortunately I cannot use you commits like this. Let me summarize what prevents me from using it:
* There are a lot of coding style violations, much of them related to using spaces where tabs are required, also line's layout got changed without sematic changes, empty lines somewhere in the nowhere
* A global static `_inval_tlb_vm` variable even used outside the scope of the "big kernel lock" in a multi-core environment is not feasible by design
* The invalidation of the VM's TLB requires a hypercall interface to the hypervisor, I hesitate to do this implicitely during world-switch, moreover now you invalidate _all_ TLB entries of all VMs
* the introduction of update_cache applies not only to stage2 entries in your implementation but to all stage1 translations too, thereby you introduce overhead for non-virtualization scenarios too
* The test cases introduced into the VMM do not only apply to our automated nightly test but to all VMM intances, e.g., when used productively in Sculpt on i.MX8, I would factor out the test cases completly, and build an own test-case VMM when it comes to test the VM API.

What I take out of this issue is that we clearly need to invalidates corresponding TLB entries when we remove a VM's mapping. As the VM interface originally did not considered memory removal, this was missing. I will leave this issue open until I found time to write  a solution therefore by introducing a clean hyper-call interface to the hypervisor.

What I still do not understand is why you introduce cache maintainance on stage2 page-table memory for the MMU. Can you please elaborate a bit more why this should be needed? From my understanding this is not needed anymore since early ARMv7 CPUs (as far as we do not change the cache attributes of the translation table memory, which we do not do).

zeehha, 2020-04-27:
@skalk thank you for your feedback. Apart from the code style violations (sorry about that sloppiness), I was well aware of the other drawbacks you pointed out. I already mentioned them at the pull-request thread [1]. 

Regarding the cache-maintenance: 
I have a use-case where I populate the s2-translation lazily, i.e. I start the Vm without Ram and attach the (super)pages on an S2-page-fault. This case requires cache-maintenance, since the created translation does not reliably reach the point-of-unification (the Vm keeps faulting on the same address, even after attaching the page). 

The Cortex-A Armv8-A Programmer's Guide [2] says: It is sometimes necessary for software to clean or invalidate a cache. [...] It can [..] be required after MMU-related activity such as changing access permissions, cache policies, or virtual to Physical Address mappings [...].

I don't know if lazy S2-translation is actually relevant for you. If this is the case, please keep me updated as It might be relevant for me and I'm curious about this topic. 

Best,
Chris

[1] https://github.com/genodelabs/genode/pull/3730
[2] https://developer.arm.com/docs/den0024/a/caches/cache-maintenance

skalk, 2020-04-27:
> The Cortex-A Armv8-A Programmer's Guide [2] says: It is sometimes necessary for software to clean or invalidate a cache. [...] It can [..] be required after MMU-related activity such as changing access permissions, cache policies, or virtual to Physical Address mappings [...].

Yes, I'm aware of the fact that there are use-cases where you have to do cache maintainance in the MMU context. But from my understanding this relates to the target memory, e.g. when a dataspace of formerly caches memory gets uncached, or similar. It should not relate to the memory containing the stage2 translations for the MMU. This is I was wondering about. In earlier ARM history, there was a need to clean cache-lines of page-table memory after changes of it, but later the MMUs could work on caches too.

To me it is relevant to really find the root cause of your observation. When I understood you right, you had empty pages-tables before, and then insert something on-demand. But you did not replaced already valid entries, right? Invalidation of the cache-lines might fix the symptom, but maybe our values in TCR_EL2 are not completely correct, which is why the MMU in stage2 relates like you observed it. If I do not make to great demands on you, would you play around with especially the fields: ORGN0, IRGN0, and SH0 in the TCR_EL2 register within `repos/base-hw/src/bootstrap/spec/arm_64/cortex_a53_mmu.cc`? Otherwise, I need to try reproduce your observation when time permits it.
In any case, thank you for your description.

zeehha, 2020-04-29:
>When I understood you right, you had empty pages-tables before, and then insert something on-demand. But you did not replaced already valid entries, right? 
Yes. I'll provide you a small POC for the issue on the weekend. If i find some additional time, I take a look at the TCR_EL2 register you mentioned.

Best,
Chris

skalk, 2020-04-29:
@zeehha Thank you very much in advance!

zeehha, 2020-05-06:
@skalk Sorry for the delay

You find the example here [1]. I wrote some comments, start with the one in `repos/os/src/server/vmm/vm.cc.` I could not find any time to try out values at TCR_EL2 though.

Best,
Chris

[1] https://github.com/zeehha/genode/tree/poc_cache_maintenance

nfeske, 2021-02-26:
@skalk What is the status of this issue?

skalk, 2022-02-17:
There is no progress on this issue, which is due to the fact that our VMM solution does not provide on-demand lazy paging of guest memory. I'll close this issue until on-demand paging becomes an issue in our virtualization solution.