genodelabs/genode #3729: vfs/File_content: fix end condition in for_each_line

rite, 2020-04-13:


rite, 2020-04-13:
e48392a fixes the occasional segmentation fault mentioned in [1]. The segmentation fault occurred on ARMv7a/base-linux when the end address of the buffer pointed to the very end of a mapped memory region. In this case `src` happend to point to the first address of an unmapped region.

[1] https://lists.genode.org/pipermail/users/2020-March/007063.html

chelmuth, 2020-04-14:
Well spotted! You're right that `src` should not be dereferenced if it points beyond the buffer. In my opinion, your current patch seems to lack support for cases where the buffer does not end with `0` or `\n`. Can you confirm this?

rite, 2020-04-14:
Indeed, good point! In this case, `fn` won't be called for the last "line"...

rite, 2020-04-14:
- 4f9c487 supports cases where the buffer does not end with `0` or `\n`
- 971cf29 removes the now redundant check whether the buffer is empty

chelmuth, 2020-04-15:
@rite many thanks for the fixups. I'll merge the complete bugfix as one commit soon.

commit 12ea4944 (2020-04-17): vfs/File_content: fix end condition in for_each_line

Fixes #3729
closed 2020-04-24 by nfeske