genodelabs/genode #3731: build.mk: duplicated entries in REPOSITORIES variable if using run tool

tomga, 2020-04-17:
After creating default linux build diractory and starting `run/log` entries in REPOSITORIES variable are duplcated. This can be observed in the following log. Value of `--repositories` argument to run tool does not have duplicated entries yet but the loop in checking library dependencies has all values twice.

```
tomga@gen:/projects/genode/genode$ tool/create_builddir linux BUILD_DIR=build/lx
Successfully created build directory at build/lx.
Please adjust build/lx/etc/build.conf according to your needs.
tomga@gen:/projects/genode/genode$ LANG=C make VERBOSE= VERBOSE_MK= VERBOSE_DIR= -C build/lx KERNEL=linux run/log
make: Entering directory '/projects/genode/genode/build/lx'
test -f "/projects/genode/genode/repos/base/run/log.run" || (echo "Error: No run script for log"; exit -1)
/projects/genode/genode/tool/run/run --genode-dir /projects/genode/genode \
                                     --name log \
                                     --specs "linux sdl x86_64 x86 64bit" \
                                     --board "" \
                                     --repositories "/projects/genode/genode/repos/base-linux /projects/genode/genode/repos/base /projects/genode/genode/repos/os /projects/genode/genode/repos/demo" \
                                     --cross-dev-prefix "/usr/local/genode/tool/19.05/bin/genode-x86-" \
                                     --qemu-args "" \

...

checking library dependencies...
\
for target in core/linux/target.mk init/target.mk test/log/target.mk; do \
  for rep in /projects/genode/genode/repos/base-linux /projects/genode/genode/repos/base /projects/genode/genode/repos/os /projects/genode/genode/repos/demo /projects/genode/genode/repos/base-linux /projects/genode/genode/repos/base /projects/genode/genode/repos/os /projects/genode/genode/repos/demo; do \
    test -f $rep/src/$target || continue; \
    make  -f /projects/genode/genode/repos/base/mk/dep_prg.mk \
            REP_DIR=$rep TARGET_MK=$rep/src/$target \
            BUILD_BASE_DIR=/projects/genode/genode/build/lx \
            SHELL=/bin/bash \
            DARK_COL="\033[00;33m" DEFAULT_COL="\033[0m" || result=false; \
    break; \
  done; \
done; $result;

...
```

It seems that it is a result of that values to REPOSITORIES variable are only appended in created by default `build.conf` file. This file is included twice in this case. Once when starting `make` from command line and recursively from run tool (called from make).

It creates a little unnecessary processing so probably has some small effect on performance (small becase I think that only string processing in make and SPECS processing are affected) but probably would be good to fix.

However I'm not sure what would be a good fix for this.

If providing REPOSITORIES variable to make from environment or command line is not supported and repositories should be modified using `build.conf` then REPOSITORIES could be initialized to empty before including `build.conf` or such reset colud be provided in default `build.conf`.

A completely safe solution would be to remove duplicates but without changing the order of entries. Following patch does this using some shell ugliness (or magic as some would argue):
```
diff --git a/tool/builddir/build.mk b/tool/builddir/build.mk
index 091c297bbd..2b523e0b0e 100644
--- a/tool/builddir/build.mk
+++ b/tool/builddir/build.mk
@@ -67,6 +67,7 @@ export LC_COLLATE=C
 # 'realpath' converts relative path names to absolute.
 #
 REPOSITORIES := $(realpath $(shell echo $(REPOSITORIES)))
+REPOSITORIES := $(shell echo $(REPOSITORIES) | sed -e 's/ /\n/g' | cat -n | sort -uk2 | sort -nk1 | cut -f2-)
 BASE_DIR     := $(realpath $(shell echo $(BASE_DIR)))
 
 #
```
but I don't even think about proposing this as a patch.

In case there is no simple cure to this issue and it will not be considered important enough then it can be closed as won't fix. It's not very important for me. I just spotted this and wanted to inform about it.

nfeske, 2020-04-22:
Thank you for bringing this issue to our attention and for the proposed fix.

You are right that the double application of the `+=` assignments is to blame. Instead of correcting the duplication after the fact, I prefer to prevent the duplication in the first place. Commit https://github.com/genodelabs/genode/commit/0fa3e01648351cc6fcbbee97e8a82099572648f4 achieves this by resetting the REPOSITORIES environment variable when invoking the run tool. This side-steps the tricky problem of removing duplicates w/o sorting in make. @tomga Do you approve?

nfeske, 2020-04-22:
I just noticed that this way of resetting the environment variable does not work as intended. I'll have to rework the commit above.

nfeske, 2020-04-22:
You can find my second attempt at https://github.com/genodelabs/genode/commit/15b002b04654716153d13cec53ed49c736e6b1c7

tomga, 2020-04-22:
@nfeske with this fix you indirectly answered my doubt that although initializing REPOSITORIES with someting like this:
```
make REPOSIOTORIES=/some/path ... 
```
worked it was only by accident and wasn't used (actually I did not test it but I'm almost sure it worked). That doubt was the reason I didn't propose a fix like in your patch.

nfeske, 2020-04-22:
The interplay of environment variables and make variables are always good for a surprise. Anyway, I think that with the second attempt (resetting the variable before including the build.conf file) we are on the safe side. Are you fine with it?

tomga, 2020-04-22:
Yes, of course. Like I initially wrote it wasn't even a problem for me. I spotted it and reported as something that probably should be fixed.

commit 0fa3e016 (2020-04-22): mk: prevent duplicates in REPOSITORIES definition

Fixes #3731
commit 15b002b0 (2020-04-22): mk: prevent duplicates in REPOSITORIES definition

Fixes #3731
commit 1459085a (2020-04-24): mk: prevent duplicates in REPOSITORIES definition

Fixes #3731
closed 2020-04-24 by nfeske