genodelabs/genode #3732: livelock during Parent::deliver_session_cap

ehmry, 2020-04-18:
I am trying to resurrect the TFTP ROM server in Genode world, but I get a livelock when delivering a second session capability. `deliver_session_cap` is called during Nic packet signal handling.

The server code is here: https://github.com/ehmry/genode-world/blob/7aa9983c9c08789b3dd2a7f7e0c823d9f2f2c606/src/server/tftp_rom/component.cc#L352

The runtime log is here: https://gist.github.com/ehmry/e6ae453f1210d301071d27a42e43d859

ehmry, 2020-04-20:
I dug around and this problem is related to init loading ld.lib.so from one of its children. Init should be able to recognize when a first generation child requests ld.lib.so for itself and forcibly route the request to its parent.

nfeske, 2020-04-20:
For the record, this is the same situation at discussed at https://github.com/genodelabs/genode/issues/2760#issuecomment-396269751. I remember that this issue actually led to https://github.com/genodelabs/genode/issues/2866.

To solve it, ld.lib.so must be routed to the parent. Granted, placing such a rule in the config is a bit inconvenient. I'm not 100% sure that this rule should be a fixed policy of init though. I'll think about it.

@ehmry May you close this issue?

ehmry, 2020-04-20:
Unfortunately it seems #2866 didn't fix this. What makes this bug frustrating is that the request for ld.lib.so is not matched by `<service name="ROM" label="ld.lib.so">`, which indicates that ld.lib.so has special treatment within init.

nfeske, 2020-04-20:
You may be able to either use `label_last="ld.lib.so"` or `unscoped_label="ld.lib.so"`. Sculpt uses the `label_last` attribute.

> What makes this bug frustrating...

You may be frustrated, which I feel sorry for. But your frustration does not make it a bug. What you call a bug, one could call a misconfiguration. I find exaggerations like that needlessly confronting.

closed 2020-04-23 by nfeske