genodelabs/genode #3736: Remove input_merger component

nfeske, 2020-04-22:
The functionality of the input merger can be achieved with the input filter using a configuration like this:

```
<config>
  <input label="ps2"/>
  <input label="usb"/>
  <output>
    <merge>
      <input name="ps2"/>
      <input name="usb"/>
    </merge>
  </output>
</config>
```

alex-ab, 2020-04-23:
Tonight all vbox scripts failed with the same symptom:
```
[2020-04-23 07:24:50] make[2]: Leaving directory '/data/autopilot/autopilot.checker.2020-04-23-01-15/x86_64/x86_64'
[2020-04-23 07:24:50] genode build completed
[2020-04-23 07:24:50] checking configuration syntax
[2020-04-23 07:24:50]   CHECK init
[2020-04-23 07:24:50] var/run/vbox5_ubuntu_14_04_32/genode/config:318: parser error : error parsing attribute name
[2020-04-23 07:24:50] 					<input_name="usb_hid"/>
[2020-04-23 07:24:50] 					           ^
[2020-04-23 07:24:50] var/run/vbox5_ubuntu_14_04_32/genode/config:318: parser error : attributes construct error
[2020-04-23 07:24:50] 					<input_name="usb_hid"/>
[2020-04-23 07:24:50] 					           ^
[2020-04-23 07:24:50] var/run/vbox5_ubuntu_14_04_32/genode/config:318: parser error : Couldn't find end of Start Tag input_name line 318
[2020-04-23 07:24:50] 					<input_name="usb_hid"/>
[2020-04-23 07:24:50] 					           ^
[2020-04-23 07:24:50] Error: Invalid XML syntax
[2020-04-23 07:24:50] Makefile:332: recipe for target 'run/vbox5_ubuntu_14_04_32' failed
```

nfeske, 2020-04-23:
Thank you very much for the notice.

nfeske, 2020-04-23:
I added the fixup 1f0238e2166ac34622f590e399e3c86349d1fdc6.

commit 623d2d99 (2020-04-22): Remove input_filter component

Fixes #3736
commit c1a87ad8 (2020-04-23): Remove input_merger component

Fixes #3736
commit 840f383e (2020-04-24): Remove input_merger component

Fixes #3736
closed 2020-04-24 by nfeske