genodelabs/genode #3742: vfs: read error handling problems

cproc, 2020-04-26:
The test case in commit https://github.com/cproc/genode/commit/0d2ac48 simulates a problem I encountered when using the lxip vfs plugin in the vfs server and an ECONNRESET error occurred while reading from a socket.

The test tries to read 1024 bytes from /dev/zero and I modified the zero vfs plugin to return READ_ERR_IO on reading. When using the zero vfs plugin locally in the test component, read() returns -1 with errno set to EIO, as expected. But when using the zero vfs plugin in the vfs server, the test component prints  `Warning: ack for unknown File_system handle 0` and the libc `read()` call does not return.

The reason for the warning is that the vfs server creates an "empty" ack packet in `Vfs_server::Io_node::_acknowledge_as_failure()`, which the client cannot correlate to the submitted packet.

When keeping `_packet` intact instead, like in `Vfs_server::Io_node::_acknowledge_as_success()`, the test component prints `Error: packet operation=0 failed` and the libc `read()` call returns 1024 read bytes, which is still not the expected result.

cproc, 2020-04-26:
Commit 9a2712e keeps the vfs server's ack packet intact on failure, but I'm not sure if this can cause other problems. Commit 7f903e0 lets the fs vfs plugin report a read IO error if the received ack packet indicates failure.

nfeske, 2020-04-27:
I wonder, would resetting the `_packet` at the end of the method work? At this point, the relevant information should be well captured in the `_acked_packet`, shouldn't it?

cproc, 2020-04-27:
Yes, that works. I changed the commit accordingly.

nfeske, 2020-04-28:
Thanks for the excellent description and the fix. I merged the latest version of your commits to staging.

nfeske, 2020-05-18:
Merged to master.

closed 2020-05-18 by nfeske