genodelabs/genode #3743: virtualbox5: enable storage device flushing

cnuke, 2020-04-28:
Virtualbox can send `flush` operations to the underlying disk storage backend. However, this is by default disabled and such request from a device module are ignored. So far it is not a problem when using Virtualbox on Sculpt because the backend works synchronously and the storage provider for virtual-disk images is normally synced periodically while raw block device access does not employ a block-cache - changes are stored persistenly in a timely manner despite no explicit `flush` operation.

That being said, I have a VFS plugin which relies on being instructed to secure changes. This is done by issuing a `sync` request on the data handle. So as a interim fix I forcefully enabled flushing in virtualbox, which addresses my use-case.

A proper evaluation is still needed when and if we want to address is.

chelmuth, 2020-05-28:
Merged to staging but wondered why we could not just undef `VBOX_IGNORE_FLUSH`.

chelmuth, 2020-05-28:
As this is work-in-discussion I removed the commit from staging again...

cnuke, 2020-06-15:
@chelmuth just undef'ing `VBOX_IGNORE_FLUSH` also works.

This, however, removes the knob for setting *ignore flush* alltogether and I'm not surce if it would be useful to keep this option around. That being said, I didn't experience any issues with my VMs so far - so I don't mind taking the undef approach.

chelmuth, 2020-06-15:
It seems we (speak you) do not need the *knob*, so I'll proceed with the undef approach.

cnuke, 2020-08-20:
@nfeske please merge 61b5c81 to staging.

nfeske, 2020-08-20:
Done.

commit c9cb678b (2020-08-20): virtualbox5: enable storage device flushing

The patches disable VBOX_IGNORE_FLUSH and a sanity check in the VMDK
backend. This enables passing an explicit flush request by the guest
down to the VFS.

Fixes #3743.
closed 2020-08-25 by nfeske