genodelabs/genode #3745: ssh client fails on Sculpt 20.02

baconicsynergy, 2020-04-29:
Hello friends!

Running Sculpt 20.02 on Lenovo Thinkpad x250 with an IP address of 192.168.0.121
My ssh server is running on Ubuntu 19.04 with an IP address of 192.168.0.140

I encountered an error when running the "ssh client" package from skalk's software repository and was hoping this was the right place to file the error and ask for help.

Per the README file in skalk's depot, I added my ssh key pair to my root directory of my primary storage as well as a "host" file containing the following: 
<host name="192.168.0.140" user="jdd" known="no"/>

When attempting to run the program, a window shows up and promptly disappears.

Here is the relevant log output from my Genode machine:

...
[runtime] child "ssh_client"
[runtime]    RAM quota: 32520K
[runtime]    cap quota: 480
[runtime]    ELF binary: init
[runtime]    priority: 0
[runtime -> ssh_client] Warning: ssh_client assigned RAM (32M) exceeds available RAM (24093621)
[runtime -> ssh_client] Warning: ssh_client assigned caps (256) exceed available caps (246)
[runtime -> ssh_client] Error: RAM preservation exceeds available memory
[runtime -> nic_router] [default] NIC sessions: 1
[runtime -> ssh_client -> ssh_client] lwIP Nic interface down
[runtime -> ssh_client -> ssh_client] libssh 0.8.4/openssl/zlib
[runtime -> ssh_client -> ssh_client] lwIP Nic interface up address=10.0.1.2 netmask=255.255.255.0 gateway=10.0.1.1
[runtime -> ssh_client -> ssh_client] 192.168.0.140 5f:e9:a9:f2:aa:fa:f8:c6:01:94:23:f6:50:c4:53:0f:8b:e2:7f:c3
[runtime -> ssh_client -> ssh_client] Error: public key authentication denied
[runtime -> ssh_client -> ssh_client] Error: password authentication denied
[runtime -> ssh_client -> ssh_client] Error: Access denied for 'none'. Authentication that can continue: publickey,password
[runtime -> ssh_client -> ssh_client] Warning: rtc not configured, returning 0
[runtime] child "ssh_client" exited with exit value -1
[runtime -> nic_router] [default] NIC sessions: 0
...

Here is the relevant log section from /var/log/auth.log on my Ubuntu server:

...
Apr 29 16:27:09 thoth sshd[3780]: Connection from 192.168.0.121 port 49152 on 192.168.0.140 port 22
Apr 29 16:27:09 thoth sshd[3780]: Accepted key RSA SHA256:xJVB6BwzYx98U/vAPNCHi6qXxw5MOXjcb6t89GoaQHQ found at /home/jdd/.ssh/authorized_keys:1
Apr 29 16:27:09 thoth sshd[3780]: Postponed publickey for jdd from 192.168.0.121 port 49152 ssh2 [preauth]
Apr 29 16:27:09 thoth sshd[3780]: Failed none for jdd from 192.168.0.121 port 49152 ssh2
...

Using ssh to remote access my servers is one of the most frequently used and important tools I use as an administrator. I'm hoping someone can help me :)

Thanks!

cnuke, 2020-05-04:
As of now the `ssh_client` component can only make use of public-keys **without** a passphrase, which is rather uncommon. It looks to me like you are trying to use a public-key that requires one as the client successfully presents the key to the server - the server happily accepts - but the client cannot unlock it and that leads to the failed login. Unfortunately the error message is somewhat sparse in that regard.

So the easiest way would be to create a new public-key w/o a passphrase to access the server, which for various reasons might not be desirable. On the other hand it should not be too hard to implement the missing passphrase query - at least in a non-interactive fashion, e.g. reading it from the *host* file directly. 

(For this reason - no clear concept of how to do it interactively - supporting passphrases was omitted IIRC.)

baconicsynergy, 2020-05-17:
Aha! It finally makes sense :)

I suppose we can keep this open until the password protected keys are usable again at some point in the future!

I really want to help but I'm only just starting to learn programming (C++ in particular). While im studying, I will continue to use Genode and search for bugs and and see if I can make myself useful!

nfeske, 2021-02-26:
There are no further steps planned for this issue. Hence, I'm closing it.

closed 2021-02-26 by nfeske