genodelabs/genode #3746: ahci: device gets stuck

alex-ab, 2020-04-30:
During stress testing (vm_stress) on X201 (current master dd899fde29448e16c96b2860c391ddcbf2880a86) it happens regularly that the AHCI device is not responding anymore. The led of the AHCI stops blinking and just stays glowing green, also the typical "sound" of the HDD goes away. After some debugging and ruling out other factors, the overall system is alive and also all components and drivers. The observation is that timeouts in the VM triggers since block requests are not arriving at all/in time and the VM fails apart.

I added a debug commit for AHCI c5f76e631e78d4d31032c5b47fcba2ca1122ef73, which regular dumps some state of the AHCI driver. The output in the case the disc LED stops working is:

```
[init -> ahci_drv] ping 4
[init -> ahci_drv] 0 READ block=608385986 count=62
[init -> ahci_drv] 1 SYNC block=0 count=224610304
[init -> ahci_drv] Warning: is=0x0 infs=0 ifs=0, ie=0x7dc000ff, serr=0x0, sdia=0xbb76
[init -> ahci_drv] ping 5
[init -> ahci_drv] 0 READ block=608385986 count=62
[init -> ahci_drv] 1 SYNC block=0 count=224610304
[init -> ahci_drv] Warning: is=0x0 infs=0 ifs=0, ie=0x7dc000ff, serr=0x0, sdia=0xbb76
[init -> seoul] VMM: BIOS disk timeout
[init -> ahci_drv] ping 6
[init -> ahci_drv] 0 READ block=608385986 count=62
[init -> ahci_drv] 1 SYNC block=0 count=224610304
```

Interestingly, ever a SYNC command is to be processed when the device gets stuck. So, I removed with 6a5a05729883116fd7c12994548909da318bbaa5 the SYNC support and now the issue did not trigger anymore during several runs. That is of course only a workaround.

Looking at https://github.com/genodelabs/genode/blob/master/repos/os/src/drivers/ahci/ata_protocol.h#L314
and 
https://github.com/genodelabs/genode/blob/master/repos/os/src/drivers/ahci/ata_protocol.h#L329
it seems that beside the sync handling, also a read command is set up, or ? In case this is intended and required, the region is quite large (and insane by the way). In the scenario the partitions in use are in the middle to end range of the hard disk, never at block 0.

@ssumpf: May you please have a look at the code and give some hints for me :-)

ssumpf, 2020-04-30:
@alex-ab: I think I see the problem:
https://github.com/genodelabs/genode/blob/dd899fde29448e16c96b2860c391ddcbf2880a86/repos/os/src/drivers/ahci/ata_protocol.h#L308
Can you try  to set the  `op.count` to `0` in case of `SYNC` and see if this solves the issue please?

chelmuth, 2020-04-30:
It seems the handling of SYNC moves to the beginning of the function step-by-step. Should we consider explicit SYNC handling upfront?

ssumpf, 2020-04-30:
@chelmuth: In case `op.count` is responsible for the issue we probably should factor it out.

alex-ab, 2020-04-30:
@ssumpf: i did so, but the device gets stuck again, ever with a read and sync to be processed. If you have an idea I can try.

ssumpf, 2020-04-30:
@alex-ab: Not really, it is probably easier to debug this issue in the office.

alex-ab, 2020-05-05:
With commit 3fff3f03bac7d648988d30be7ad871b24773c9b0 (+ 566671103a166b56df88d42702f169caf2e9f7c6) the issue goes away for me, thanks @ssumpf.

nfeske, 2020-05-07:
Cool! Thank you for the investigation. I merged both commits to staging.

commit 7fb3c116 (2020-05-07): ahci: SYNC with empty NCQ queue only

Fixes #3746
commit ea78d85d (2020-05-18): ahci: SYNC with empty NCQ queue only

Fixes #3746
closed 2020-05-18 by nfeske