genodelabs/genode #3750: Remove `Allocator_guard`

nfeske, 2020-05-05:
The `Allocator_guard` is not water-tight as described in #1039. For guarding the resource consumption, there exists the `Constrained_ram_allocator`, which works reliably but at the granularity of dataspaces. So there exists a clean solution but the flaky `Allocator_guard` still exists also.

We should finally retire the old `Allocator_guard` now. However, this requires some work because the utility is still used in a number of places, most notably core, nitpicker, and the platform driver.

nfeske, 2020-05-06:
Commit https://github.com/genodelabs/genode/commit/f045e7c3a3503c389870037b78618a2f132b7fd0 removes the `Allocator_guard` from all components except for the Intel GPU driver. The latter, I better don't touch because this change would require a quite elaborate refactoring and the need for thorough manual testing. I'd defer this to the time when we pick up the practical work on GPU driver the next time. For now, I have placed a copy of the allocator_guard.h local to the component. @cnuke is that ok from your perspective?

cnuke, 2020-05-07:
@nfeske that's perfectly fine.

commit f045e7c3 (2020-05-06): Remove Allocator_guard

This patch removes old 'Allocator_guard' utility and replaces its use
with the modern 'Constrained_ram_allocator'.

The adjustment of core in this respect has the side effect of a more
accurate capability accounting in core's CPU, TRACE, and RM services.
In particular, the dataspace capabilities needed for core-internal
allocations via the 'Sliced_heap' are accounted to the client now.
The same goes for nitpicker and nic_dump as other former users of the
allocator guard. Hence, the patch also touches code at the client and
server sides related to these services.

The on [... cut]
commit 9d67f9fc (2020-05-18): Remove Allocator_guard

This patch removes old 'Allocator_guard' utility and replaces its use
with the modern 'Constrained_ram_allocator'.

The adjustment of core in this respect has the side effect of a more
accurate capability accounting in core's CPU, TRACE, and RM services.
In particular, the dataspace capabilities needed for core-internal
allocations via the 'Sliced_heap' are accounted to the client now.
The same goes for nitpicker and nic_dump as other former users of the
allocator guard. Hence, the patch also touches code at the client and
server sides related to these services.

The on [... cut]
closed 2020-05-18 by nfeske