genodelabs/genode #3755: Deprecate unsafe `Xml_node` methods

nfeske, 2020-05-06:
We should wrap up  #3125 by marking the old methods as deprecated, and removing them in the subsequent release cycle.

nfeske, 2020-05-07:
The commit https://github.com/nfeske/genode/commit/ec615a61cd957c577e2e72655c9ebb277ba4b221 is necessary to build solo5 until the port is updated. @ehmry Can you take care of adjusting the use of the `Xml_node` in the solo5 code to avoid the deprecated methods? With the updated port, we can revert the commit.

ehmry, 2020-05-14:
I believe the XML accessors have already been updated in the bindings - https://github.com/Solo5/solo5/blob/master/bindings/genode/bindings.cc.

ehmry, 2020-05-14:
There was a new release today, so its a good time for someone who understands the ports to make an update. https://github.com/Solo5/solo5/releases/tag/v0.6.5

nfeske, 2020-05-14:
My intermediate attempt is at https://github.com/nfeske/genode/commit/f1a6d2cac046ac94d214d92018e8abc81fb7074e. I had to make a few adjustments (a new header, new use of _Static_assert). Still it is not there yet.

```
In file included from /plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/../bindings.h:33,
                 from /plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:42:
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/include/solo5/mft_abi.h:157:41: error: types may not be defined within __builtin_offsetof
 #define MFT1_NOTE_ALIGN offsetof(struct { char c; struct mft m; }, m)
                                         ^
/usr/local/genode/tool/19.05/bin/../lib/gcc/x86_64-pc-elf/8.3.0/include/stddef.h:417:52: note: in definition of macro ‘offsetof’
 #define offsetof(TYPE, MEMBER) __builtin_offsetof (TYPE, MEMBER)
                                                    ^~~~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/include/solo5/mft_abi.h:159:50: note: in expansion of macro ‘MFT1_NOTE_ALIGN’
 _Static_assert((offsetof(struct mft1_note, m) & (MFT1_NOTE_ALIGN - 1)) == 0,
                                                  ^~~~~~~~~~~~~~~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc: In member function ‘solo5_result_t Solo5::Platform::net_acquire(const char*, solo5_handle_t&, solo5_net_info&)’:
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:474:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (me != nullptr && index < MFT_MAX_ENTRIES)
   ^~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:476:4: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
    return devices[index]->net_info(info);
    ^~~~~~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc: In function ‘void Component::construct(Genode::Env&)’:
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:635:28: error: ‘res’ was not declared in this scope
   Genode::error("Solo5: ", res, " Built-in manifest validation failed. Aborting.");
                            ^~~
```

nfeske, 2020-05-14:
Better taking Solo5 to a dedicated issue: #3761

commit 42dd0503 (2020-05-11): Xml_node: mark deprecated methods

Fixes #3755
commit dd0c1575 (2020-05-18): Xml_node: mark deprecated methods

Fixes #3755
closed 2020-05-18 by nfeske