genodelabs/genode #3756: depot_query segfaults while processing a runtime file

trimpim, 2020-05-07:
After a rebuild of all my the depot archives suddenly depot_query crashes (Segmentation fault (signum=11)) during processing the deploy config (platform is `base-linux` on `x86_64`).

Runing the run script with strace via `strace -e trace=mmap,munmap,clone -ff -o /tmp/debug/pid make -C build/linux run/<runscript>` shows, that a memory access to the first byte of a page that was previously `munmap`ed causes a SIGSEGV.

As this reminded me of the Issue #3729, I ensured that this patch is applied to my current branch.

Attaching `gdb` to `depot_query` enables me to extract the following information:

 - the message when the error is triggered
   ```
   (gdb) c
   Continuing.

   Thread 2 "ld.lib.so" received signal SIGSEGV, Segmentation fault.
   0x000000000100933f in Genode::Token<Genode::Xml_attribute::Scanner_policy_xml_identifier>::_calc_len (this=0x401fdb10, max_len=1) at /home/pirmin/GapFruit/genode/repos/base/include/util/string.h:278
   278			return (c == '\t' || c == ' ' || c == '\n');
   ```

 - the current object
   ```
   (gdb) p *this
   $2 = {_start = 0x7faeb2306fff "\n"<error: Cannot access memory at address 0x7faeb2307000>, _max_len = 1, _len = 1075829712}
   ```

 - the backtrace
   ```
   (gdb) bt
   #0  0x000000000100933f in Genode::Token<Genode::Xml_attribute::Scanner_policy_xml_identifier>::_calc_len (this=0x401fdb10, max_len=1) at /home/pirmin/GapFruit/genode/repos/base/include/util/string.h:278
   #1  0x000000000100957e in Genode::Token<Genode::Xml_attribute::Scanner_policy_xml_identifier>::Token (max_len=1, s=<optimized out>, this=0x401fdb10) at /home/pirmin/GapFruit/genode/repos/base/include/util/token.h:108
   #2  Genode::Token<Genode::Xml_attribute::Scanner_policy_xml_identifier>::next (this=<synthetic pointer>) at /home/pirmin/GapFruit/genode/repos/base/include/util/token.h:108
   #3  Genode::Xml_node::Tag::next_token (this=this@entry=0x401fe5e0) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_node.h:416
   #4  0x0000000001017d53 in Genode::Xml_node::with_raw_node<Depot_query::Main::_query_blueprint(const Path&, Genode::Xml_generator&)::<lambda(Genode::Xml_node)>::<lambda()>::<lambda(char const*, Genode::size_t)> > (fn=..., this=0x401fe590) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_node.h:761
   #5  Depot_query::Main::<lambda(Genode::Xml_node)>::<lambda()>::operator() (__closure=<optimized out>) at /home/pirmin/GapFruit/genode/repos/gems/src/app/depot_query/main.cc:647
   #6  Genode::Xml_generator::Node::Node<Depot_query::Main::_query_blueprint(const Path&, Genode::Xml_generator&)::<lambda(Genode::Xml_node)>::<lambda()> > (name=0x1023c2b "pkg", func=..., xml=..., this=0x401fdb70) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_generator.h:290
   #7  Genode::Xml_generator::node<Depot_query::Main::_query_blueprint(const Path&, Genode::Xml_generator&)::<lambda(Genode::Xml_node)>::<lambda()> > (func=..., name=0x1023c2b "pkg", this=0x401ff0e0) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_generator.h:349
   #8  Depot_query::Main::<lambda(Genode::Xml_node)>::operator()(Genode::Xml_node) const (__closure=<optimized out>, node=...) at /home/pirmin/GapFruit/genode/repos/gems/src/app/depot_query/main.cc:592
   #9  0x0000000001018c24 in Genode::File_content::xml<Depot_query::Main::_query_blueprint(const Path&, Genode::Xml_generator&)::<lambda(Genode::Xml_node)> > (this=0x401fe640, this=0x401fe640, fn=...)
    at /home/pirmin/GapFruit/genode/repos/base/include/util/noncopyable.h:52
   #10 Depot_query::Main::_query_blueprint (this=0x102bdc0 <Component::construct(Genode::Env&)::main>, pkg_path=..., xml=...) at /home/pirmin/GapFruit/genode/repos/gems/src/app/depot_query/main.cc:589
   #11 0x0000000001018d79 in Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}::operator()(Genode::Xml_generator&) const::{lambda(Genode::Xml_node)#1}::operator()({lambda(Genode::Xml_generator&)#2}) const (node=..., 
    __closure=0x401fed90) at /home/pirmin/GapFruit/genode/repos/base/include/util/interface.h:20
   #12 Genode::Xml_node::for_each_sub_node<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}::operator()(Genode::Xml_generator&) const::{lambda(Genode::Xml_node)#1}>(char const*, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}::operator()(Genode::Xml_generator&) const::{lambda(Genode::Xml_node)#1} const&) const (this=this@entry=0x401fefc0, type=type@entry=0x1023eec "blueprint", fn=...) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_node.h:962
   #13 0x000000000101aaf4 in Genode::Xml_node::for_each_sub_node<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}::operator()(Genode::Xml_generator&) const::{lambda(Genode::Xml_node)#1}>(char const*, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}::operator()(Genode::Xml_generator&) const::{lambda(Genode::Xml_node)#1} const&) const (fn=..., type=0x1023eec "blueprint", this=0x401fefc0) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_node.h:955
   #14 Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}::operator()(Genode::Xml_generator&) const (__closure=<optimized out>, __closure=<optimized out>, xml=...)
    at /home/pirmin/GapFruit/genode/repos/gems/src/app/depot_query/main.cc:467
   #15 Depot_query::Main::_gen_versioned_report<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}>(Genode::Constructible<Genode::Expanding_reporter>&, Genode::String<32ul> const&, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2} const&)::{lambda(Genode::Xml_generator&)#1}::operator()(Genode::Xml_generator&) const (xml=..., this=<optimized out>) at /home/pirmin/GapFruit/genode/repos/gems/src/app/depot_query/main.cc:383
   #16 Genode::Expanding_reporter::generate<Depot_query::Main::_gen_versioned_report<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}>(Genode::Constructible<Genode::Expanding_reporter>&, Genode::String<32ul> const&, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2} const&)::{lambda(Genode::Xml_generator&)#1}>(Depot_query::Main::_gen_versioned_report<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}>(Genode::Constructible<Genode::Expanding_reporter>&, Genode::String<32ul> const&, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2} const&)::{lambda(Genode::Xml_generator&)#1} const)::{lambda()#1}::operator()() const::{lambda()#1}::operator()() const (
    this=<optimized out>, this=<optimized out>) at /home/pirmin/GapFruit/genode/repos/os/include/os/reporter.h:198
   #17 Genode::Xml_generator::Node::Node<Genode::Expanding_reporter::generate<Depot_query::Main::_gen_versioned_report<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}>(Genode::Constructible<Genode::Expanding_reporter>&, Genode::String<32ul> const&, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2} const&)::{lambda(Genode::Xml_generator&)#1}>(Depot_query::Main::_gen_versioned_report<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}>(Genode::Constructible<Genode::Expanding_reporter>&, Genode::String<32ul> const&, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2} const&)::{lambda(Genode::Xml_generator&)#1} const)::{lambda()#1}::operator()() const::{lambda()#1}> (
    func=..., name=<optimized out>, xml=..., this=0x401feda0) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_generator.h:290
   #18 Genode::Xml_generator::node<Genode::Expanding_reporter::generate<Depot_query::Main::_gen_versioned_report<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}>(Genode::Constructible<Genode::Expanding_reporter>&, Genode::String<32ul> const&, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2} const&)::{lambda(Genode::Xml_generator&)#1}>(Depot_query::Main::_gen_versioned_report<Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2}>(Genode::Constructible<Genode::Expanding_reporter>&, Genode::String<32ul> const&, Depot_query::Main::_handle_config()::{lambda(Genode::Xml_generator&)#2} const&)::{lambda(Genode::Xml_generator&)#1} const)::{lambda()#1}::operator()() const::{lambda()#1}> (
    func=..., name=<optimized out>, this=0x401ff0e0) at /home/pirmin/GapFruit/genode/repos/base/include/util/xml_generator.h:349
[... cut ...]
   #26 0x00000000010215f7 in Depot_query::Main::Main (this=0x102bdc0 <Component::construct(Genode::Env&)::main>, env=...) at /home/pirmin/GapFruit/genode/repos/gems/src/app/depot_query/main.cc:514
   #27 0x000000000100007e in Component::construct (env=...) at /home/pirmin/GapFruit/genode/repos/gems/src/app/depot_query/main.cc:802
   ```

 - the runtime file that is parsed is form the `chroot` package (which hasn't changed)

If understand the code correctly, the tokenizer of the XML parser tries to see if there is an other token and steps over the page boundary as the last byte of the runtime file happens to be exactly the last byte of the page it lies in memory.

nfeske, 2020-05-07:
Yes, you are right! Thank you for uncovering this issue. The `Xml_node` takes the `max_len` as argument but this information is apparently not propagated all the way down to the tokenizing level. The code down here assumes that the string is null-terminated.

nfeske, 2020-05-07:
> The code down here assumes that the string is null-terminated.

I was not correct with my impulse response above. All the information is actually nicely passed down to the tokenizing code. However, I think I spotted the (quite subtle) problem within the `Token::_calc_len` method. @trimpim May you give commit https://github.com/nfeske/genode/commit/a19c2c7e2cd57c140236c3742f1942cd79ca55ca a try?

nfeske, 2020-05-07:
I forgot to mention, your bug report is superb! Thank you very much for all the context and the backtraces.

trimpim, 2020-05-07:
@nfeske With your patch I can run my previously broken branch again. Thanks for this very quick fix for the issue.

nfeske, 2020-05-08:
I added https://github.com/genodelabs/genode/commit/6b784da71256c9f2a7ac3fa754baa37d89161358 to staging, which contains a fix for a closely related issue along with a regression test.

commit 6b784da7 (2020-05-08): util/token.h: fix possible out-of-bounds read

The 'WHITESPACE' case of the _calc_len method wrongly accessed the
character before checking upper bound of the token. The problem is fixed
by switching the order of both conditions.

Fixes #3756
commit 48b24568 (2020-05-18): util/token.h: fix possible out-of-bounds read

The 'WHITESPACE' case of the _calc_len method wrongly accessed the
character before checking upper bound of the token. The problem is fixed
by switching the order of both conditions.

Fixes #3756
closed 2020-05-18 by nfeske