genodelabs/genode #3757: dde_bsd: audio recording regression introduced with recent driver update

cnuke, 2020-05-07:
The driver update (commit 9d7a58f6a7a8b085a5650b42fc32c696a9058b21) breaks recording on at least the Fujitsu Lifebook S938. Reverting the commit restores the functionality.

cnuke, 2020-05-09:
As it turned out, just the explicit `<mixer field="record.enable" value="on"/>` was missing, which I already hoped would be the sole reason for the none working recording functionality. Since I neglected testing it when updating the driver I feared it might have other reasons. That being said, I successfully tested recording on
  * a Shuttle DS57U using a headset
  * a Thinkpad X220 as well as on
  * a Fujitsu Lifebook S938

both using internal speakers/external headphone and the internal microphone. Using the internal speakers and micrphone might lead to some nasty feedback loops (as is the case when using the `audio_in.run` test) but apart from that the quality is decent.

(Unfortunately, I could not make use of an external microphone on both laptops as I was not able to configure the codec widgets properly. I tried the same using my native OpenBSD 6.6 installation on the X220 without success, so it could be a general probleme - PEBKAC not withstanding…)

Commit 929e4a9 extends the documention while 046183e removes the service toggle attributes.

nfeske, 2020-05-11:
Thank you for the welcome update. I merged both commits to staging.

commit b6590fa9 (2020-05-11): dde_bsd: remove configration attributes

The driver always opens the audio device in duplex mode, i.e.,
playback and recording. Setting the 'playback' or 'recording' attribute
only influnces the service announcement. Due to changes made in a more
recent OpenBSD release recording must be set enabled explicitly anyway.
Since we already provide the interface that mirrors the one used by
OpenBSD 1:1 in the configuration use that and the remove the additional
config attributes.

Fixes #3757.
commit f1427fc0 (2020-05-18): dde_bsd: remove configration attributes

The driver always opens the audio device in duplex mode, i.e.,
playback and recording. Setting the 'playback' or 'recording' attribute
only influnces the service announcement. Due to changes made in a more
recent OpenBSD release recording must be set enabled explicitly anyway.
Since we already provide the interface that mirrors the one used by
OpenBSD 1:1 in the configuration use that and the remove the additional
config attributes.

Fixes #3757.
closed 2020-05-18 by nfeske