genodelabs/genode #3761: Update Solo5 to version 0.6.5

nfeske, 2020-05-14:
As pointed out by @ehmry, a new version of Solo5 has become available: https://github.com/Solo5/solo5/releases/tag/v0.6.5

nfeske, 2020-05-14:
My intermediate attempt of an update can be found at https://github.com/nfeske/genode/commit/f1a6d2cac046ac94d214d92018e8abc81fb7074e. I had to make a few adjustments (a new header, new use of _Static_assert). Still it is not there yet.

```
In file included from /plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/../bindings.h:33,
                 from /plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:42:
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/include/solo5/mft_abi.h:157:41: error: types may not be defined within __builtin_offsetof
 #define MFT1_NOTE_ALIGN offsetof(struct { char c; struct mft m; }, m)
                                         ^
/usr/local/genode/tool/19.05/bin/../lib/gcc/x86_64-pc-elf/8.3.0/include/stddef.h:417:52: note: in definition of macro ‘offsetof’
 #define offsetof(TYPE, MEMBER) __builtin_offsetof (TYPE, MEMBER)
                                                    ^~~~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/include/solo5/mft_abi.h:159:50: note: in expansion of macro ‘MFT1_NOTE_ALIGN’
 _Static_assert((offsetof(struct mft1_note, m) & (MFT1_NOTE_ALIGN - 1)) == 0,
                                                  ^~~~~~~~~~~~~~~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc: In member function ‘solo5_result_t Solo5::Platform::net_acquire(const char*, solo5_handle_t&, solo5_net_info&)’:
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:474:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (me != nullptr && index < MFT_MAX_ENTRIES)
   ^~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:476:4: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
    return devices[index]->net_info(info);
    ^~~~~~
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc: In function ‘void Component::construct(Genode::Env&)’:
/plain/no/genode.git/contrib/solo5-ffd2d5b283030ad04551176c482401fb702430d8/src/lib/solo5/bindings/genode/bindings.cc:635:28: error: ‘res’ was not declared in this scope
   Genode::error("Solo5: ", res, " Built-in manifest validation failed. Aborting.");
                            ^~~
```

ehmry, 2020-05-15:
The timer test of 0.6.5 fails, but there is a patch https://github.com/Solo5/solo5/pull/466.

nfeske, 2020-05-18:
@ehmry Do you have a hint about how to go about the "error: types may not be defined within __builtin_offsetof" build error I posted above? I must admit that the  `offsetoff` magic is alien to me.

ehmry, 2020-05-25:
I think the problem might be that this code is valid C but not C++. I have a patch that fixes both of these isssues:
https://git.sr.ht/~ehmry/genodepkgs/tree/master/packages/solo5/genode.patch#L82
https://git.sr.ht/~ehmry/genodepkgs/tree/master/packages/solo5/genode.patch#L298

nfeske, 2020-05-26:
Thanks @ehmry for the links to your patches. I have added your patch in commit e3a6c9861d468860fb8180311d257cb8ae0eb73b.

Unfortunately, there seem to remain other build issues:

```
/.../src/lib/solo5/tests/test_time/../../bindings/printf.c:530:26: error: ‘intmax_t’ undeclared (first use in this function); did you mean ‘int64_t’?
         flags |= (sizeof(intmax_t) == sizeof(long) ? FLAGS_LONG : FLAGS_LONG_LONG);
                          ^~~~~~~~
                          int64_t
```
and
```
/usr/local/genode/tool/19.05/bin/../lib/gcc/x86_64-pc-elf/8.3.0/../../../../x86_64-pc-elf/bin/ld: test_blk.o: in function `solo5_app_main':
/.../src/lib/solo5/tests/test_blk/test_blk.c:60: undefined reference to `solo5_block_acquire'
```

ehmry, 2020-05-26:
I have a fixup at 72edf5d1fe6a06abe0b3427c97612d716936186f that make everything compile again. There is an outstanding problem, the tests will fail without an embedded manifest listing the devices to open session for, and the tool for embedding the manifest must be built from the Solo5 sources.

https://github.com/Solo5/solo5/blob/master/tests/Makefile.tests
https://github.com/Solo5/solo5/blob/master/tests/test_blk/manifest.json

nfeske, 2020-06-26:
I renamed the issue to "Move Solo5 to genode-world" because the solo5 runtime support is not actively maintained by anyone at Genode Labs. The version that is currently tested by our CI is out of date by now.

commit 8bb59391 (2020-06-26): Move Solo5 to genode-world

Fixes #3761
commit 5249f335 (2020-06-29): Move Solo5 to genode-world

Fixes #3761
closed 2020-06-29 by nfeske