genodelabs/genode #3763: Pipe producer gets stuck on pipe close

cnuke, 2020-05-15:
While using a pipe to trim some input I encountered a problem that appeared to be a memory leak. The problem can be reproduced by executing the following command using the `bash.run` script:

```
dd if=/dev/zero bs=8194 count=1 | dd bs=1 count=1
```

Doing that the producing **dd** gets stuck in the libc in the `_vfs_sync` call in `close()`.

The following LOG snippts contains the results of executing

```
dd if=/dev/zero bs=8193 count=1 | dd bs=1 count=1
dd if=/dev/zero bs=8194 count=1 | dd bs=1 count=1
dd if=/dev/zero bs=8194 count=1 | dd bs=1 count=1
```

```
[init -> /bin/bash]   0x40000000 .. 0x4fffffff: stack area
[init -> /bin/bash]   0x50000000 .. 0x60217fff: ld.lib.so
[init -> /bin/bash]   0xadb8000 .. 0xaffffff: libc.lib.so
[init -> /bin/bash]   0xad0c000 .. 0xadb7fff: vfs.lib.so
[init -> /bin/bash]   0xaccf000 .. 0xad0bfff: libm.lib.so
[init -> /bin/bash]   0xacbe000 .. 0xaccefff: posix.lib.so
[init -> /bin/bash] __sys_fork: res: 1
[init -> /bin/bash -> 1] __sys_fork: res: 0
[init -> /bin/bash -> 1] filename: '/bin/dd'
[init -> /bin/bash -> 1] close:508: fd_path: '/dev/pipe/0/in'
[init -> /bin/bash] __sys_fork: res: 2
[init -> /bin/bash -> 2] __sys_fork: res: 0
[init -> /bin/bash -> 2] filename: '/bin/dd'
[init -> /bin/bash -> 1] close:510: fd_path: '/dev/pipe/0/in'
[init -> /bin/bash -> 2] close:508: fd_path: '/dev/terminal'
[init -> /bin/bash -> 1] close:508: fd_path: '/dev/terminal'
[init -> /bin/bash -> 1] close:510: fd_path: '/dev/terminal'
[init -> /bin/bash] exit: pid: 1 exit_code: 0
[init -> /bin/bash] _handle_exit: pid: 1 exit_code: 0
[init -> /bin/bash] exiting pid: 1 avail: 22755864 used: 2896K
Warning: blocking canceled in entrypoint constructor
[init -> /bin/bash -> 2] close:510: fd_path: '/dev/terminal'
[init -> /bin/bash -> 2] close:508: fd_path: '/dev/terminal'
[init -> /bin/bash -> 2] close:510: fd_path: '/dev/terminal'
[init -> /bin/bash] exit: pid: 2 exit_code: 0
[init -> /bin/bash] exiting pid: 2 avail: 25998906 used: 2368K
Warning: invalid acknowledgement for control message
[init -> /bin/bash] __sys_fork: res: 3
[init -> /bin/bash -> 3] __sys_fork: res: 0
[init -> /bin/bash -> 3] filename: '/bin/dd'
[init -> /bin/bash -> 3] close:508: fd_path: '/dev/pipe/0/in'
[init -> /bin/bash] __sys_fork: res: 4
[init -> /bin/bash -> 4] __sys_fork: res: 0
[init -> /bin/bash -> 4] filename: '/bin/dd'
[init -> /bin/bash -> 4] close:508: fd_path: '/dev/terminal'
[init -> /bin/bash -> 4] close:510: fd_path: '/dev/terminal'
[init -> /bin/bash -> 4] close:508: fd_path: '/dev/terminal'
[init -> /bin/bash -> 4] close:510: fd_path: '/dev/terminal'
[init -> /bin/bash] exit: pid: 4 exit_code: 0
[init -> /bin/bash] _handle_exit: pid: 4 exit_code: 0
[init -> /bin/bash] exiting pid: 4 avail: 22755864 used: 2896K
[init -> /bin/bash] __sys_fork: res: 5
[init -> /bin/bash -> 5] __sys_fork: res: 0
[init -> /bin/bash -> 5] filename: '/bin/dd'
[init -> /bin/bash -> 5] close:508: fd_path: '/dev/pipe/1/in'
[init -> /bin/bash] __sys_fork: res: 6
[init -> /bin/bash -> 6] __sys_fork: res: 0
[init -> /bin/bash -> 6] filename: '/bin/dd'
[init -> /bin/bash -> 6] close:508: fd_path: '/dev/terminal'
[init -> /bin/bash -> 6] close:510: fd_path: '/dev/terminal'
[init -> /bin/bash -> 6] close:508: fd_path: '/dev/terminal'
[init -> /bin/bash -> 6] close:510: fd_path: '/dev/terminal'
[init -> /bin/bash] exit: pid: 6 exit_code: 0
[init -> /bin/bash] _handle_exit: pid: 6 exit_code: 0
[init -> /bin/bash] exiting pid: 6 avail: 19508726 used: 3428K
```

to illustrate the issue.

nfeske, 2023-09-12:
Today I tried to reproduce this issue given the steps above. I'm unable to observe the symptoms. Granted, quite a few VFS internals changed in the meantime.

@cnuke is this issue still relevant or can it be closed?

cnuke, 2023-09-12:
@nfeske unfortunately I cannot recall the context in which I encountered this problem and since it appears to be fine now I am all for closing this issue.

closed 2023-09-12 by cnuke