genodelabs/genode #3764: autopilot: log-file-creation problem

nfeske, 2020-05-19:
When executing the example given at https://genode.org/documentation/release-notes/19.08#Build_system_and_tools, I get the following error:

```
$ ./tool/autopilot -t x86_64-pc-sel4 -t x86_64-pc-nova -r run/log
genode dir    : /plain/no/genode.git
targets       : x86_64-pc-nova x86_64-pc-sel4
architectures : x86_64
run scripts   : run/log
test dir      : /tmp/autopilot.no
make -j       : 2

--- architecture x86_64 ---
pc nova:                       run/log                        couldn't open "/tmp/autopilot.no/x86_64.pc.nova.run/log.log": no such file or directory
    while executing
"open [log_file $arch $board $kernel $run_script] "WRONLY CREAT TRUNC""
    (procedure "log_fd" line 9)
    invoked from within
"log_fd $arch $board $kernel $run_script"
    (procedure "execute_run_script" line 4)
    invoked from within
"execute_run_script $arch $board $kernel $run_script"
    ("foreach" body line 10)
    invoked from within
"foreach target $targets {
			if { [get_arch $target] ne $arch } { continue }

			set board  [get_board $target]
			set kernel [get_kernel $target]

		..."
    ("foreach" body line 3)
    invoked from within
"foreach run_script $run_scripts {

		foreach target $targets {
			if { [get_arch $target] ne $arch } { continue }

			set board  [get_board $target]
	..."
    ("foreach" body line 5)
    invoked from within
"foreach arch $architectures {

	puts stderr "\n--- architecture $arch ---"

	foreach run_script $run_scripts {

		foreach target $targets {
			if { [g..."
    (file "./tool/autopilot" line 450)
```

I noticed this while updating the Genode Foundations book. I suspect that this problem is related the changed name of the build directory. The code for creating the log file assumes the directory being named "/tmp/autopilot.no/x86_64.pc.nova.run/" but I only have the a directory named after the architecture ("/tmp/autopilot.no/x86_64/").

@chelmuth What do you think would be the right way to go about it?

chelmuth, 2020-05-19:
I tested `./tool/autopilot -t x86_64-pc-linux -r log` which works, while you used `-r run/log`. Could this be the error?

nfeske, 2020-05-19:
It is! Thank you for spotting the problem so quickly. Commit https://github.com/genodelabs/genode/commit/528e51242957e35a581f7d7d268a52dd326838cc corrects the commands of the release notes.

chelmuth, 2020-05-19:
It's a pity as I wrote the section of the release notes... thanks for the fix.

commit 528e5124 (2020-05-19): doc: fix autopilot command in release notes 19.08

Fixes #3764
commit 128ba651 (2020-05-27): doc: fix autopilot command in release notes 19.08

Fixes #3764
closed 2020-05-27 by chelmuth