genodelabs/genode #3765: lwip: assertion pbuf_free after running 20 tcp connections in parallel

sidhussmann, 2020-05-22:
This issue is a result of a discussion on the mailing list [0].

The problem can be reproduced as follows:
  netcat.sh:
```bash
    #!/bin/bash
    for i in $(seq 10000); do
      dd if=/dev/zero status=none count=4 | netcat 10.0.2.55 8899 > /dev/null
    done
```
```bash
  > parallel -j 20 ./netcat.sh -- $(seq 20)
```

I also removed the diagnostic messages and after 422 rounds I got the
following log messages.

```log
[init -> test-tcp_echo_server] waiting for connection 422
[init -> test-tcp_echo_server] Error: tcp_err_callback arg=null
[init -> test-tcp_echo_server] Error: Assertion "pbuf_free: p->ref > 0" /plain/krishna/src/genode/genode_tmp.git/contrib/lwip-6e0661b21fde397041389d5d8db906b5a6543700/src/lib/lwip/src/core/pbuf.c:753
[init -> test-tcp_echo_server] Error: tcp_poll_callback
[init -> test-tcp_echo_server] Error: tcp_poll_callback socket=653982970 state=4
[init -> test-tcp_echo_server] Error: tcp_poll_callback local_port=8899 remote_port=34522 state=4
[init -> test-tcp_echo_server] Error: tcp_poll_callback
```

This assertion seems to be very serious and the code decrements the
reference counter also if the assertion fails, which should render
this pbuf not reclaimable for a long time due to the integer
wrap-around. In my test case the lwip stack stills happily responds to
ARP and ICMP ping.

[0] https://lists.genode.org/pipermail/users/2020-May/007158.html

chelmuth, 2020-05-22:
I can reproduce both #3765 and #3766 with the parallel netcat instances.

chelmuth, 2020-07-30:
I can no longer reproduce this issue when applying monitored VFS library processing from my WiP branch (see commit 5797b826dbbb0c476882a6518a83fe79bab88f56).

chelmuth, 2020-10-09:
@sidhussmann can we close this issue?

sidhussmann, 2020-10-09:
@chelmuth yes. Thank you so much for your effort!

closed 2020-10-09 by chelmuth