genodelabs/genode #3766: lwip: deadlock in combination with pthread

sidhussmann, 2020-05-22:
This issue is a result of a discussion on the mailing list [0].

[0] https://lists.genode.org/pipermail/users/2020-May/007158.html

sidhussmann, 2020-05-22:
@chelmuth wrote:
> It's a pity that the most important information is hidden in the LWIP
> data structures like: Is there pbuf memory pressure? How long are the
> send queues? Are there incoming packets? Do ARP and ICMP still work?

After the "deadlock" occurs, the lwIP stack does not respond to ARP nor ICMP. However, I noticed in wireshark [0] that lwIP is still trying to re-transmit a packet a couple of times, even though the client has closed the connection with a RST (Port 8899 -> 38868). 
I then implemented the `Lwip::tcp_poll_callback()` in f1eb563 which gets called to notify the application (in this case vfs) when a connection is idle. After "deadlock", the `tcp_poll_callback` function gets called repeatedly forever. So the lwIP stack does not seem to be completely stuck. However, the RST from the client has not been handled at all.

1. I'm trying to figure out how a pthread maps to data being sent via lwIP. E.g. when calling `write()` from a pthread is it the same thread in which lwIP is sending the data? If not, is it safe to ignore the hints from [1]?
2. In `Lwip::tcp_sent_callback()` f1eb563 we ignore the length argument completely. Is this intended and handled with the `process_io` functionality?
3. I didn't find an easy way to find out more about queues of lwIP. With my current findings, does it still make sense to iterate through all the linked lists of lwIP?
4. Since lwIP still handles callbacks and re-transmitts packets after the deadlock, the problem seems to be `vfs` related. Would you agree @chelmuth?

[0] https://cloud.gapfruit.com/s/BeGEZjw9qFMizbd
[1] http://www.nongnu.org/lwip/2_1_x/multithreading.html

sidhussmann, 2020-05-22:
> https://github.com/genodelabs/genode/issues/3765#issuecomment-632710302

@chelmuth 
I'm curious how you reproduced this issue using netcat

chelmuth, 2020-05-22:
As I wrote: I ran the test and also ran netcat.sh with parallel. Sometimes it triggers #3765, more often it deadlocks like described in this issue.

sidhussmann, 2020-05-22:
However, using `netcat` as you described, does not result in `tcp_poll_callback()` being called.

chelmuth, 2020-07-30:
I merged the five commits above to our staging branch and are no longer able to produce lwip deadlocks with the following test cases.

- Using netcat as documented in #3765 
- `ab -c 10 -n 10 http://10.0.2.55:8899/`
- `siege -c 10 -r 10 -d 0 10.0.2.55:8899`
- `nping -c 50 --tcp-connect --data-length 600 -rate 10 -p 8899 10.0.2.55`

Please note, due to the implementation of the tcp-echo test program it may block in a resource request on high concurrency or rate of connections.

chelmuth, 2020-10-09:
@sidhussmann can we close this issue?

sidhussmann, 2020-10-09:
@chelmuth yes. Thank you so much for your effort!

closed 2020-10-09 by chelmuth