genodelabs/genode #3767: quota-safe pattern for connections opened on behalf of a client

skalk, 2020-05-25:
During the work on a new platform_drv design for ARM (issue #3299), I stumbled upon the problem of opening connections on behalf of a client. Namely, per-device of a platform client IO_MEM or IRQ sessions are opened and delegated to the client. Especially, in the case of the IO_MEM session, the costs are not always known in advance accurately. We can only assume them in a pessimistic way to not endanger the platform_drv running out of resources.
When opening a connection to a service the Genode::Connection and Genode::Env interplay hides whether more resources than the initially spent ones are needed to successfully open up the connection.
My proposal would be to introduce another Genode::Env class that does not try to make additional payment, but escalates the Quota_exceeded exceptions. This pattern might be useful for all servers that need to open up sessions on behalf of client sessions.

nfeske, 2020-05-25:
This reminds me of the GDB monitor, which has the same problem with intercepting the debugging target's PD and CPU sessions.

nfeske, 2021-10-08:
Commit https://github.com/genodelabs/genode/commit/172577594299098509370932ad4c812d9101d4f0 addresses a closely related problem where init implicitly paid for session-quota upgrades at session-creation time instead of reflecting the condition to the session's client.