genodelabs/genode #3768: dynamic_rom_session: Quota_exceeded should not be hidden

skalk, 2020-05-25:
During the work on the platform_drv for ARM (issue #3299), I hit a problem in the Dynamic_rom_session abstraction. While using a Constrained_ram_allocator used per client session as allocator argument for the Dynamic_rom_session, either a Out_of_caps or Out_of_ram exception was caught in the Dynamic_rom_session::_unsynchronized_update routine, which results in an invalid dataspace without any sign to the caller.
In my eyes, it would be better to escalate the exceptions to the environment, so that abstractions like the Constrained_ram_allocator, session quota upgrades can fulfill their job.

nfeske, 2020-06-22:
This can currently not be avoided because the ROM session does not reflect an `Out_of_ram` or `Out_of_caps` exception to its client. After the session creation, the session is expected to just work. If the server happens to require additional resources during the lifetime of the session, the server must pay.

This could arguably be changed in the future. But such an interface change goes well beyond the scope of this issue and must be carefully planned.

I'm wondering how we should go about this issue? It cannot be solved in isolation. Should it stay open as a reminder or should we close it?

skalk, 2020-06-22:
I'm aware that this implies a fundamental change including API changes, and did not want to act precipitately. Therefore, it was meant as a reminder for future changes. I know myself to be forgetful regarding such kind of issues, but if you are confident to keep it in mind, we can also close it again.