genodelabs/genode #3773: Update OpenSSL to version 1.1.1

chelmuth, 2020-06-04:
This issue collects necessary work items to the most recent stable OpenSSL version (currently 1.1.1g).

* [X] Update port to 1.1.1 (for x86-64 only)
* [X] Adapt/update all libssl/libcrypto users in *genode* repo
* [ ] Extend support to x86-32 and ARM
* [ ] Use libssl.map/libcrypto.map version scripts to reduce ABI (also adapt symbol files)
* [ ] Adapt all libssl/libcrypto users in *world* repo

chelmuth, 2020-06-23:
Fetchurl tests work fine for x86 now. ARM file lists, assembler generation and linking works. Unfortunately, the library startup probes CPU features by using the desired instruction and catching `SIGILL` in *armcap.c*. As we do not have support for this signal in our libc yet, the file must be patched in a sane way (by using the most portable implementation).

chelmuth, 2020-06-23:
3ef4801a2f3787560a429082a71904f9d86a7b95 forces NEON instructions on armv7/8 and effectively shortcuts the dynamic probing. Now, the test also succeeds on pbxa9, next imx8 and rpi.

trimpim, 2020-12-18:
@chelmuth I have started porting your patches to current master. I will push my feature branch as soon I could run some checks, to verify it works.

trimpim, 2020-12-23:
@chelmuth I have updated openssl to 1.1.1i. With some small fixes I was able to run `run/lighttpd` on the imx8q_evk as well as on x86_64.

For our main scenario I had to disable NEON, as the SoC used in it doesn't support NEON. I currently only have commented out the line enabling NEON. This patch surely needs some refinement. Do you think that openssl with NEON support should be the default on ARM platforms?

In `https://github.com/trimpim/genode-world/tree/openssl11-20.11` I have collected the changes I had to make in `genode-world`, in order to build everything in our automated CI-pipeline. Most of this is not run-tested yet.

trimpim, 2021-01-04:
@chelmuth the symbols file of curl already only exports the symbols from `libcurl.ver`.

chelmuth, 2021-01-04:
> For our main scenario I had to disable NEON, as the SoC used in it doesn't support NEON. I currently only have commented out the line enabling NEON. This patch surely needs some refinement. Do you think that openssl with NEON support should be the default on ARM platforms?

I can remember the reason for forcing NEON was that otherwise the library probes for CPU features at runtime and the probing code needs support for SIGILL, which is not supported currently. I tested with Qemu/pbxa9.

trimpim, 2021-01-05:
@chelmuth if I execute `run/fetchurl_lwip` for `base-hw` on `pbxa9/quemu` with my tree, The run script is successful.

Which run script did you run to test your version of openssl?

It could be, that this changed form version `1.1.1g` to `1.1.1i`.

chelmuth, 2021-01-05:
I slept over your suggestion and think it is quite good. So, we should not use NEON per default on ARMv7. Additionally, we should provide the option to define `SPECS += neon` when building libcrypto for ARMv7. On ARMv8, I tend to use NEON by default accroding to https://en.wikipedia.org/wiki/AArch64

> To both AArch32 and AArch64, ARMv8-A makes VFPv3/v4 and advanced SIMD (Neon) standard.

IMO, the implementation does not need the original *armcap.c* source file anymore as we could just implement `void OPENSSL_cpuid_setup(void)` and `uint32_t OPENSSL_rdtsc(void)` for NEON and not-NEON.

trimpim, 2021-02-18:
@chelmuth as I started preparations to integrate the upcoming Genode 21.02, I created a branch on top of current master that contains a consolidated set of patches. I kept the NEON/non-NEN changes and the reduction of exported symbols as separate patches for easier review.

Regarding the reduction of symbols was your indention to create something automatic so that it could be used in a future update? I did it manually via multiple `grep` commands.

https://github.com/trimpim/genode-world/tree/openssl11-pre-21.02 contains my changes to `genode-world`. As we do not use them except for gRPC they are not tested except that they compile.

chelmuth, 2021-02-18:
I merged your commits to staging and also adapted genode-world/staging. Did you have any issues with `make run/fetchurl_lwip`? My one-shot test on `KERNEL=linux` did not work but `make run/depot_download` succeeded.

trimpim, 2021-02-18:
@chelmuth I did not run `make run/fetchurl_lwip` but I will look in to it in the next days.

trimpim, 2021-02-19:
@chelmuth for me `run/fetchurl_lwip` with `KERNEL=linux` for me also fails on current `master`.

chelmuth, 2021-02-19:
@trimpim my bad, fetchurl is not tested on linux for autopilot. I have small fixes for last nights hiccups and are looking forward to the weekend results.

chelmuth, 2021-02-26:
This issue was fixed in Genode 21.02.

closed 2021-02-26 by chelmuth