genodelabs/genode #3775: pthread: support to configure placement strategy

alex-ab, 2020-06-05:


nfeske, 2020-06-08:
This looks very good to me. @alex-ab do you think that it should go to staging, or do you have reservations (not setting the fixed label yet)?

nfeske, 2020-06-08:
Should we consider making `!single-cpu` the default?

alex-ab, 2020-06-08:
> Should we consider making !single-cpu the default?

I have no fixed opinion about that. I tend to say yes, I would prefer.

On the other side, with that we would change all pthread applications to run on several CPUs, which was not the case beforehand. We may just argue, that we document it with the next release this switch. E.g. - no pthread configuration means to use all available CPUs according to the affinity Genode configuration of the start node.

alex-ab, 2020-06-08:
After some discussion during lunch, I'm not certain about whether this should be the default handling. Opinions were quite diverse, what should be default and what not. Maybe we should require/obligate a pthread configuration ever so that the intent is documented by the integrator, which would remove any uncertainty. This of course would require to adjust all scenarios.

nfeske, 2020-06-08:
> Maybe we should require/obligate a pthread configuration [...]

I think that this goes too far because I expect that most commonly, the intention is "I don't care or don't know". Unless there is a security risk with the default, we should not bother the user in the common case.

I see three arguments in favor of making the use of all (reachable) CPUs default:

* It triggers concurrency bugs earlier than later,
* It improves performance in the "don't care" case, which I assume is the common one,
* It is limited to the affinity configuration imposed on the subsystem

Note that I wouldn't necessarily apply the same arguments to `fork` because fork/execve is often used in plain sequential ways where cross-CPU-talk may slow things down. But for `pthreads` I think that it would be fine to assume that the scaling to multiple CPUs is most often the intent behind using threads.

What are the arguments in favor of mapping all `pthreads` to only a single CPU by default?

alex-ab, 2020-06-08:
> What are the arguments in favor of mapping all pthreads to only a single CPU by default?

I have none beside legacy behavior.

I added another commit which implements as default behavior the usage of all CPUs. If anybody else has more special requirements, he may come up with a follow up commit.

alex-ab, 2020-06-08:
> This looks very good to me. @alex-ab do you think that it should go to staging, or do you have reservations (not setting the fixed label yet)?

I'm up for staging now

nfeske, 2020-06-09:
Merged to staging.

BTW, @alex-ab have you tested building virtualbox with the patch?

alex-ab, 2020-06-09:
> BTW, @alex-ab have you tested building virtualbox with the patch?

Yes.

commit dc98d14d (2020-06-09): libc: support pthread placement configuration

Up to now all pthreads get placed on one CPU.

The patch adds support to evaluate a libc pthread configuration specifying
a placement strategy.

The default strategy is now to use all available CPUs as imposed by Genode's
affinity configuration for the pthread application.

The old behaviour, putting all pthreads on one single cpu can be still
configured by:

<libc>
 <pthread placement="single-cpu"/>
 ...
</libc>

Fixes #3775
commit 87b08d6c (2020-06-22): libc: support pthread placement configuration

Up to now all pthreads get placed on one CPU.

The patch adds support to evaluate a libc pthread configuration specifying
a placement strategy.

The default strategy is now to use all available CPUs as imposed by Genode's
affinity configuration for the pthread application.

The old behaviour, putting all pthreads on one single cpu can be still
configured by:

<libc>
 <pthread placement="single-cpu"/>
 ...
</libc>

Fixes #3775
closed 2020-06-22 by nfeske