genodelabs/genode #3779: ldso: library loaded via 'Shared_object' API does not find symbol of other API-loaded library

cproc, 2020-06-13:
Commit 0d3f3b6 enhances the 'Shared_objetct' API ldso test so that two shared libraries are loaded via this API now. The main program calls a test function in each of the two libraries. The test function of the first API-loaded library calls a function in a test library which was loaded at program start time, which works. The test function of the second API-loaded library calls the same function as the first one, which works, and in addition a function of the first API-loaded library, which unfortunately doesn't work:

```
[init -> test -> test-ldso] Shared-object API
[init -> test -> test-ldso] -----------------
[init -> test -> test-ldso] Global object constructed
[init -> test -> test-ldso] lib_dl_1_test: call 'lib_1_good':
[init -> test -> test-ldso] lib_1_good
[init -> test -> test-ldso] lib_dl_2_test: call 'lib_1_good':
[init -> test -> test-ldso] lib_1_good
[init -> test -> test-ldso] lib_dl_2_test: call 'lib_dl_1_good':
[init -> test -> test-ldso] Error: LD: jump slot relocation failed for symbol: 'lib_dl_1_good()'
```

The problem originally occurred when I converted the Qt5 libraries implementing 'QNitpickerViewWidget' and 'QPluginWidget' to 'dlopen'ed Qt plugins, because they depend on the now also 'dlopen'ed QPA plugin.

chelmuth, 2020-06-15:
Isn't this expected behavior? If a shared object is loaded by `dlopen` the exported symbols are not added to any lookup table implicitly. So, if `lib_dl_2_test` needs `lib_dl_1_good` the symbol must be looked up explicitly like for `lib_dl_1_test`, for example

```c++
typedef void (*lib_dl_1_good_t)();
lib_dl_1_good_t lib_dl_1_good = (lib_dl_1_good_t)lib_dl_1_so.lookup("lib_dl_1_good");
```

chelmuth, 2020-06-15:
Commit 17df52bfbd21655b024cb4c3a34318f71cd3e60a removed the warning in `dlopen` if the `RTLD_GLOBAL` flag is specified but does not add the feature to implicitly add the exported symbols to the lookup table. @ssumpf do you agree?

ssumpf, 2020-06-15:
Our dynamic linker currently does not implement proper `RTLD_GLOBAL` semantics, but rather `RTLD_LOCAL`.

chelmuth, 2020-06-15:
Good start for documentation of RTLD semantics is https://pubs.opengroup.org/onlinepubs/9699919799/functions/dlopen.html. All dependency resolution and relocation mechanics/orders are specified there. The pages also contain information about how `dlclose()` can be implemented.

nfeske, 2021-02-26:
Reading this issue, I'm wondering. Will we live with this limitation or are there any steps planned to eliminate it? In any case, I'd not label this issue as bug because it is merely a missing feature.

chelmuth, 2023-06-01:
The issue got rather stale and I can't see any ambitions to implement `RTLD_GLOBAL`. So, I added a warning about unsupported RTLD_GLOBAL in dlopen() in commit 5e03ff709fce937e66446833eb67736f4763d330 (marked as fix).

@cproc @ssumpf if the additional noise is not wanted, we could remove the commit again and just close the issue.

ssumpf, 2023-06-01:
> The issue got rather stale and I can't see any ambitions to implement `RTLD_GLOBAL`. So, I added a warning about unsupported RTLD_GLOBAL in dlopen() in commit [5e03ff7](https://github.com/genodelabs/genode/commit/5e03ff709fce937e66446833eb67736f4763d330) (marked as fix).
> 
> @cproc @ssumpf if the additional noise is not wanted, we could remove the commit again and just close the issue.

@chelmuth: Sounds good to me. As we haven't had a use case were `RTLD_GLOBAL` has been a strict requirement - meaning a scenario did work , I would suggest to not implement this feature prematurely.

commit 5e03ff70 (2023-06-01): libc: warn about unsupported RTLD_GLOBAL in dlopen()

Fixes #3779
commit 4a15621d (2023-06-16): libc: warn about unsupported RTLD_GLOBAL in dlopen()

Fixes #3779
closed 2023-06-16 by nfeske