genodelabs/genode #3781: os: Block session server backed by VFS library

cnuke, 2020-06-15:


nfeske, 2020-06-16:
That looks very cool. With this component in place, could we remove the existing rom_block and ram_block servers?

cnuke, 2020-06-16:
@nfeske thanks, I considered this as well.

So far the `vfs_block` component needs to know the length of the underlying file to calculate the block count. To replace `ram_block` we would need a memory-backed file; the `ram` VFS-plugin is a natural fit. Unfortunately we cannot specify the total length of a file in the `ram` file system. That
leaves as with (at least) two ways forward:

* Add a *size* attribute to the `vfs_block` policy that sets the expected size of the block device. In case it is specified we omit querying the underyling file for its length (or could issue a warning if it differs).

* Extend the `ram` VFS-plugin so that one can set the total length of a file.

Out of those options I prefer the first one as this would also enable us to grow the underlying file,
i.e, start with a small (sparse) file that gets large by time and is limited in length by the `vfs_block` policy.

I'll prepare a fixup commit if that's to your liking.

nfeske, 2020-06-16:
Alternatively, we could use a `<ram>` file system and create the file with the `<import>` plugin, which populates the VFS at initialization time. This leaves the question of how to "import" a synthetic file of a given size. For this, we could consider adding a `size` attribute the the `<inline/>` file system. When specified, the inline file would be inflated to the specified size. To illustrate the idea:

```
<vfs>
  <ram/>
  <import>
    <inline size="200M"/>
  </import>
</vfs>
```

For the rom_block replacement, simply mounting a `<rom>` would fit the bill, wouldn't it?

cnuke, 2020-06-16:
> Alternatively, we could use a `<ram>` file system and create the file with the `<import>` plugin, which populates the VFS at initialization time. This leaves the question of how to "import" a [...]

That's an interesting idea and would work as well. The down-side is having to change the `<inline>` file system as it is currently tied to the `Xml_node`. Of course that would be justified if there are other use-cases where an artificially sized `<inline>` file is needed. At least in the `vfs_block` use-case this approach would increase the start-up time due to the copy operation, which arguably is neglectable.

Either way, independent of how we address the synthetic file problem, handling the expected *size* at the `vfs_block` component as a feature sounds reasonable to me.

> For the rom_block replacement, simply mounting a `<rom>` would fit the bill, wouldn't it?

Yes, it should.

cnuke, 2020-06-16:
Ah nevermind, handling the expected *size* at the `vfs_block` could get complicated - in case of a writeable session doing the seek and performing the needed write first is fine but for obvious reasons that does not work on reads. So requiring the underlying file to have a proper size is probably more reasonable after all.

nfeske, 2020-06-17:
> The down-side is having to change the <inline> file system as it is currently tied to the Xml_node

While looking at the commit https://github.com/cnuke/genode/commit/0fb828b7b4b530bfac62b380778c04ce697b18ce, I have to agree. Your patch illustrates well that this feature is an unnatural addition to the inline fs. I'm sorry for misguiding you in this direction. It would be cleaner to introduce a dedicated file-system type for this case. E.g., how would you think about a `<zeros size="10M"/>` read-only file system? In practice, it would probably be used by the `<import/> fs only. Of course, I'm open for other names but I couldn't think of a better one.

cnuke, 2020-06-18:
Yes, I also had this feeling but I hestitated to introduce a new file system because I was not sure how to name it properly.

There is already the `<zero>` file system used for providing `/dev/zero` and additionally having `<zeros>` might add confusion. Maybe adding a `capacity` attribute to the `<zero>` file system could work (although that also feels somewhat unnatural as well).

chelmuth, 2020-06-18:
Adding an optional `size` or `capacity` attribute to `<zero>` does not feel unnatural to me - it seems ot be quite handy.

cnuke, 2020-06-18:
I went with @chelmuth's suggestion and added a `size` attribute to the `<zero>` fs and I am now in the process of replacing `rom_block` and `ram_block`, which tackles a few more run scripts than anticipated.

nfeske, 2020-06-18:
Thank you very much for jumping through all those hoops!

cnuke, 2020-06-19:
The following commit series contains the discussed changes and also replaces `ram_block` and `rom_block`.

6c2280670e os: remove rom_block server component
2358266a6d Replace 'rom_block' with 'vfs_block'
317234db86 os: remove ram_block server component
e7079c2ac0 Replace 'ram_block' with 'vfs_block'
fbce23da9c depot_autopilot: test case for vfs_block
02c634e4a1 os: add test-vfs_block pkg recipe
3213c4abec os: Block session server backed by VFS library
129a0a19be vfs: extend <zero> fs to limit amount of zeros
55f49bd531 block_tester: limit batching in sequential test
16ce6aa4f7 os: add block_tester src recipe

nfeske, 2020-06-22:
Thank you! I merged your commit series to staging.

cnuke, 2020-06-23:
@nfeske @chelmuth please merge af19046, 8482e7c and dee4f19 to fix CI failures.

nfeske, 2020-06-23:
You are welcome.

cnuke, 2020-06-25:
@chelmuth @nfeske please merge 0b829da to staging.

nfeske, 2020-06-25:
Thanks. I merged your fixup.

commit e1aab829 (2020-06-22): os: Block session server backed by VFS library

The 'vfs_block' component will give access to a VFS file through a
Block session. For more detailed information please look at its
README.

(On a technical note, the server currently only allows for one
active session and has only one pending back end request but can
easily be extended in the future.)

Fixes #3781.
closed 2020-06-29 by nfeske