genodelabs/genode #3782: nitpicker: page fault during Session_component::realloc_buffer

alex-ab, 2020-06-17:
Rarely I see a page fault in nitpicker, whenever a client stops (scrcpy) working. I could track it down to the mentioned method of nitpicker:

```
[init -> scrcpy] INFO: 0 fps
[init -> scrcpy] sdl event 32770
[init -> scrcpy] WARN: Device disconnected
[init -> scrcpy] CRITICAL: Requested to kill -1, this is an error. Please report the bug.
[init -> scrcpy] Warning: raise: raise not implemented
[init -> scrcpy] Warning: raise: raise not implemented
[init -> nitpicker] Error:  ? 0x0 size=0x0
no RM attachment (READ pf_addr=0x40 pf_ip=0x1023fac from pager_object: pd='init -> nitpicker' thread='ep') 
Warning: page fault, pager_object: pd='init -> nitpicker' thread='ep' ip=0x1023fac fault-addr=0x40 type=no-page
```

```
--- a/repos/os/src/server/nitpicker/session_component.cc
+++ b/repos/os/src/server/nitpicker/session_component.cc
@@ -515,6 +515,8 @@ Buffer *Session_component::realloc_buffer(Framebuffer::Mode mode, bool use_alpha
        /* copy old buffer content into new buffer and release old buffer */
        if (src_texture) {
 
+               Genode::error(" ? ", texture, " size=", size);
+
                Surface<PT> surface(texture->pixel(),
                                    texture->Texture_base::size());
 
```

nfeske, 2020-06-17:
Thank you for reporting!

From your instrumentation, it is clear that the code does not consider `texture` becoming a `nullptr`, which definitely needs fixing. If `try_alloc_texture` returns a `nullptr`, we should back out early while resetting the values of `_buffer_size`, `_texture`, `_uses_alpha`, and `_input_mask`.

That bug notwithstanding, I'd like to get a better understanding of the situation because it may hint at a potential performance problem. The `realloc_buffer` method is executed whenever a client's virtual framebuffer changes its size. It is costly. I wonder why is it executed during the normal lifetime of scrcpy?

alex-ab, 2020-06-17:
In this case it is executed once during sdl2 cleanup/shutdown code, not during the normal lifetime. It happens that the code reduce the dataspace to width/height 0

nfeske, 2020-06-17:
> It happens that the code reduce the gui fb to width/height 0

That makes perfect sense.

commit 03155a2b (2020-07-09): nitpicker: handle failed try_alloc_texture

Fixes #3782
closed 2020-07-13 by nfeske