genodelabs/genode #3783: hw: acknowledge IRQs directly at the kernel where possible

skalk, 2020-06-17:
On platforms where interrupts are not shared across different devices, we do not have to acknowledge the handled interrupt at the IRQ service, e.g. provided by platform driver or core. Instead, we can directly acknowledge them at the kernel. When working on issue #3247 I measured up to 20% cpu consumption by the irq thread in core when forwarding all 4000 interrupts/sec of the broken dwc_otg usb device in the rpi1. This consumption was due to the acknowledgement IPC to core.

The only platform with interrupt sharing (not using MSI/MSI-X) that base-hw supports is x86 AFAIK. So, it should be feasible to implement the Irq_session_client dependent on the kernel and architecture, so that it normally acknowledges the interrupt handling at the IRQ service, but not in the base-library implementation for base-hw on architectures different than x86, where it directly calls the kernel. A WIP commit showed a significant performance win for the pathologic case of the rpi1 broken usb controller.

tomga, 2020-07-01:
> A WIP commit showed a significant performance win for the pathologic case of the rpi1 broken usb controller.

@skalk have you published this commit somewhere?

skalk, 2020-07-02:
@tomga it is commit id 5936386a8b125a5f748e344f67840a8bab898419 in my Genode repository.

tomga, 2020-07-02:
@skalk Thanks for pushing it now. I'll try to test its effect on my branch.