genodelabs/genode #3784: Change pixel format to 32-bit RGB888

nfeske, 2020-06-17:
Genode's restriction to 16-bit color depth got repeatedly questioned in the past (#2045). We should finally leave this legacy behind and move the entire GUI stack to the use of 32-bits per pixel.

I'm not going to introduce a new option but completely replace the old format by the new one in one swift change. This is because there is no sensible reason to let the GUI server use a color mode less that 32-bit RGB888. If we left the choice to the applications, we would increase the complexity of the entire GUI stack (need for negotiation, pixel-format conversion). By settling globally on one format, we keep the drivers, GUI multiplexers, widget-set back-ends as low complex as possible.

The alpha channel will be handled as a distinct plane that is separate from the color values. That means that we leave one byte of every pixel unused. The rationale behind this decision is given here: https://lists.genode.org/pipermail/users/2018-March/005976.html

nfeske, 2020-06-22:
I merged https://github.com/genodelabs/genode/commit/c81a379c7d4ac86260435b2be25494a84be8989d to staging despite two three remaining problems/unknowns:

- Seoul does not longer display the guest framebuffer because I failed to trick the guest to set up a 32-bit VESA mode. @alex-ab I'd be grateful if you can have a look at this issue. The interfacing with the Genode GUI seems though. At least, one can see the textual boot messages.
- I updated and tested almost all framebuffer drivers. However, I haven't run the i.MX8 framebuffer driver on the EVK board. @skalk Could you give it a spin?
- Similarly, I updated the boot_fb_drv but failed to test it in Qemu.

alex-ab, 2020-06-23:
> * Seoul does not longer display the guest framebuffer because I failed to trick the guest to set up a 32-bit VESA mode. @alex-ab I'd be grateful if you can have a look at this issue. The interfacing

fe54de438ae7f2f03aa12946203920c83ca18a1f works for me

> * Similarly, I updated the boot_fb_drv but failed to test it in Qemu.

Works for me in Qemu.

chelmuth, 2020-06-23:
> > * Similarly, I updated the boot_fb_drv but failed to test it in Qemu.
> 
> Works for me in Qemu.

Nice! May you share your test procedure for the record here? Thanks.

nfeske, 2020-06-23:
@alex-ab Thank you! I added your fixup to staging.

alex-ab, 2020-06-23:
> Nice! May you share your test procedure for the record here? Thank

I adjusted framebuffer.run, 305996c77f6ef2d69535adf526a9a85b35dd3b36, and used image/uefi and used an oldish Qemu 2.4.

nfeske, 2020-06-23:
On the i.MX8 EVK, the demo.run test failed with
```
[init -> drivers] child "fb_drv" requests resources: ram_quota=2101248[0m[0m
```

Apparently, we'd need to adjust the quota in https://github.com/genodelabs/genode/blob/dfcb6109b57ba8636e5796de227ced2af04315c3/repos/dde_linux/recipes/raw/drivers_interactive-imx8q_evk/drivers.config#L59 but I'm not sure by how much. @skalk could you experiment a bit with the board?

chelmuth, 2020-06-23:
> I adjusted framebuffer.run, [305996c](https://github.com/genodelabs/genode/commit/305996c77f6ef2d69535adf526a9a85b35dd3b36), and used image/uefi and used an **oldish Qemu 2.4.**

I think the bold part is what I was missing. Maybe you told me before, but what's the reason for `image/uef`i not working in newer Qemu versions?

alex-ab, 2020-06-23:
> I think the bold part is what I was missing. Maybe you told me before, but what's the reason for `image/uefi` not working in newer Qemu versions?

I don't know the reason, I just know the fact that our tianocore firmware works for some version of Qemu. I mainly remembered the question was on the mailing list [1](https://lists.genode.org/pipermail/users/2018-October/006429.html) and [2](https://lists.genode.org/pipermail/users/2018-October/006430.html)

alex-ab, 2020-06-23:
@chelmuth, @nfeske: may you please try commit 9f9e3db74a7a76c624d07e29cd36df56e2ed8d9f with image/uefi and add to staging, if it works out for you

nfeske, 2020-06-23:
@alex-ab That's nice. The less binary data we host, the better.

nfeske, 2020-06-29:
The change to the 32-bit pixel format is complete, including the adaptation of the components in the genode-world repository.

commit c81a379c (2020-06-22): Change pixel format to 32 bits per pixel

Fixes #3784
commit ef741ef8 (2020-06-29): Change pixel format to 32 bits per pixel

Until now, Genode's framebuffer session interface was based on the
RGB565 pixel format. This patch changes the pixel format to 32-bit
XRGB where the X part is ignored. It adapts all graphical applications
and device drivers accordingly.

The patch also adjusts the users of the drivers_interactive packages,
assigning 64 MiB RAM and 1500 caps to the drivers subsystem, which is
sufficient for covering high resolutions at 32 bits per pixel and to
accommodate multi-component USB HID input stacks.

Fixes #3784
closed 2020-06-29 by nfeske